ArduPilot/ardupilot #3298: single/coax heli and Copter motor library conflict..

AndKe, 2015-12-07:
AP_MotorsMulticopter.h  contains descriptions for MOT_SPIN_ARMED and many more.
This is not included in http://copter.ardupilot.com/wiki/arducopter-parameters/#mot__parameters  - which only conains other MOT_parameters. from AP_MotorsSingle.h or AP_MotorsCoax.h

APM Planner 2 and (other GCS) is also missing the parameter descriptions because of this export problem.

R-Lefebvre, 2015-12-08:
That's strange.  Why would the parser only be pulling from single/coax, and not multicopter?

One issue is that single/coax code is not being maintained by anybody and falling behind.  IMO, it should be deprecated unless somebody is willing to take it on.

splenkingforth, 2016-01-20:
I'm doing some work with single code so would rather it was not deprecated though time poor right now.

IamPete1, 2021-07-25:
Looks like this has been fixed.

closed 2021-07-25 by IamPete1