ArduPilot/ardupilot #3303: AP3.4: Switch from MANU to GUIDED stops climbs and takes a while to climb to safe altitude

marcmerlin, 2015-12-08:
This is reproducible 100%:
1) take off in MANU, high throttle, high pitch up, gain altitude fast
2) flip toggle switch to enable geofencing, which instantly triggers due to a geofence minimum altitude limit of 70m

Everytime, geofencing, first stops the climb, then turns, potentially towards trees, and then slowly resumes the climb. 
This kind of worries me a bit. Why is it not aggressively trying to get back to the minimum altitude deck first? Isn't the minimum altitude deck the first priority (i.e. not hitting trees?)

Example: https://youtu.be/CHkpbfeCJo4?t=22s

actually it's worse than that, watch at offset 1:15, guided mode actually pitches _down_ and descends while turning into unknown terrain before finally applying 90% throttle and doing an aggressive climb again.
I do not have a lateral box (well, I do, it's the size of the US), so shouldn't it continue the climb it was already in and slowly make its way to the RTL point instead of turning aggressively, pitching down, and losing altitude before finally climbing again? 

http://marc.merlins.org/tmp/MANU_to_GUIDED_4.BIN

magicrub, 2015-12-08:
I suspect the stall prevention kicked in.
Check TRIM_ARSPD_CM and ARSPD_FBW_MIN.

The OSD shows the AS/GS (Air/Gnd speed) being 39 (10.8 m/s) at the point
you switch from Manual to guided and it raises to 78 (22 m/s) before it
goes away. Since guided is an auto mode it will try and maintain a certain
TRIM_ARSPD_CM airspeed via throttle. When STALL_PREVENTION=1 then
ARSPD_FBW_MIN is used as a threshold where when below that airspeed it also
pitches down to increase airspeed.

marcmerlin, 2015-12-08:
Mmmh, you're right, I did have 12m/s as minimum airspeed which is not the real stall speed, but because I'm getting bad misbehaviours in https://github.com/diydrones/ardupilot/issues/3032  I raised it to 12m/s (and after my last report, just raised it 14m/s because I'm scared of having video die, and then RTL go into that oscillation of death mode and me not being able to take over to fly it back without the video once RTL has stopped working).

So you're right, this is a byproduct of the other issue and my stall speed being set lower than necessary, but two possible improvements though:
1) it should have a different stall speed for no motor vs full motor. Many RC planes can fly way below their stall speed on motor power
2) even as is, the nose down is too aggressive and then it takes too long to get back from pitch down (12 degrees down, already doing 69kph) back to pitch up.

If you think either is worth looking into as an improvement, we can keep this open. If not, feel free to close it.

magicrub, 2015-12-08:
I think the issue was the case of it switching between unmanaged and
managed airspeed. Just hand launch in auto! :)

marcmerlin, 2015-12-08:
Auto is good if I have a destination in mind, and I program it for each location. I could do this, but currently it's easier to just launch manual, switch to guided, put FPV goggles on, and then FBWA or CRUISE for manual flight :)
But back to this issue, is 1) or 2) worth considering and actionable? If not, we can close this issue. What's going on at this point is not a bug per se, but could behave better as explained in the last reply.

magicrub, 2015-12-08:
Give this a try: upload this mission and launch in AUTO:
- TAKEOFF, param1 is fixed pitch, alt is what to climb to. Heading is
locked to whatever direction you launched it at.
- LOITER_UNLIMITED, all params zero mean just loiter forever where you're
at.

From there you can easily hand launch and loiter until you're fpv-ready
without having to give it any gps lat/lng. That allows you to use that
single mission at any location.

marcmerlin, 2015-12-20:
Hi @magicrub, I didn't get to try auto takeoff on my last set of test flights and now my motor is died, so it'll be a little while, but I'll try your recommendation.
Either way, something seems wrong with GUIDED, I have a more clear example in bug #3357 where GUIDED almost flew my plane into the ground.
I'll close this bug in favour of #3357

closed 2015-12-20 by marcmerlin