ArduPilot/ardupilot #3308: Copter should not use GPS for LAND

imrj, 2015-12-08:
based on this thread and its associated incident (https://github.com/diydrones/ardupilot/issues/3286#issuecomment-162381578)

maybe as a configurable parameter or a fixed setting, copter should not rely on GPS for auto LAND procedure, as nearby objects and/or building obstructions could affect GPS signal and cause unexpected changes in position during a land operation.  People and/or property are relatively closer to the copter when its landing and more exposed, which could result on injury or damage if a sudden and unexpected position change occurs due to relying on poor GPS signal at very low altitudes.
The copter should explicitly rely on other methods, such as EKF or INS? to maintain its position during a landing, rather than a fluctuating or poor GPS signal.
or maybe also this setting can be triggered when the copter enters the final landing hover (which is 10m or so I believe) so stop relying on GPS then? just suggesting enhancements based on limited knowledge to avoid potentially dangerous situation when copter is very close to people

govsux, 2015-12-08:
Without the gps I think it would potentially drift far more than any inaccuracies with the gps. Without the gps it has no idea where it is right? Also I think it's the compass that can really get 'off' when getting close to buildings / wires / metal etc. No?? But perhaps there is a way....

magicrub, 2015-12-08:
The problem on that issue that you linked was due to a dual-GPS
configuration which is not recommended.

magicrub, 2015-12-08:
I vote we close this but will leave it for @rmackay9 to decide.

imrj, 2015-12-08:
magicrub...the issue could have happened with a single GPS...its a totally valid scenario, that even with a single GPS, the signal could degrade due to obstructions at low altitude  and lead to a rapid shift due to position error resulting in a dangerous condition.  As a matter of fact in my example , it was not the GPS switch in itself that caused the position error, it was just a bad GPS condition after that...

magicrub, 2015-12-08:
Turning off GPS is a bad GPS condition. The flight controller doesn't
magically navigate and hold its position without GPS. The solution is not
to disable GPS, you'll need to rely on some other sensor hardware such as:

IRLock:
http://copter.ardupilot.com/wiki/precision-landing-with-irlock/

Optical flow:
http://copter.ardupilot.com/wiki/common-optical-flow-sensors-landingpage/

imrj, 2015-12-08:
but this happens today already right? because EKF failsafe does a LAND as an option, so surely its doing it w/o relying on GPS.   I am no expert but I would think when the LAND command is issued the copter just needs to "lock on" to a position and stay there using internal navigation only? is that possible?  does the land under a failed EKF condition is then prone to drifting off?

WickedShell, 2015-12-09:
My hardware not in flight has a drift of 6m/s when sitting on my desk right
now. I've seen higher numbers right now, I'm not sure how you could expect
that to be better when landing? In general GPS will be far better then not
having it, and as Tom said you need a different hardware solution if you
want to avoid GPS.

govsux, 2015-12-09:
@imrj  This seems on the way so seems like things are headed in the right direction. Seems like in the future if it one could select in params RTL or RTM via gps until IR lock is acquired then disregard gps
http://copter.ardupilot.com/wiki/precision-landing-with-irlock/

imrj, 2015-12-09:
thats good govsux....I am just merely putting a recommendation out there for anything that can be done to avoid a potentially dangerous condition where the copter acts unexpectedly and with little time for corrective action. To me is worrisome  to have the copter doing a rather peaceful maneuver like a landing and then all the sudden and w/o any warning just shoot off wildly and crash.....
Yes i know this could be as untreatable just as a motor dying or some other component failing, but if we know about it (ie. bad GPS position, etc) we should do something about it perhaps, to the extent possible of course.

rmackay9, 2015-12-09:
To clear up misunderstandings though, the incident linked above was most likely caused by a mechanical failure, it's almost certainly not a GPS issue.

I'm going to close this issue if that's ok because it's really requesting that the vehicle maintain a vertical descent but not use the GPS which isn't possible without a replacement like IRLock or Optical flow.

closed 2015-12-09 by rmackay9

imrj, 2015-12-09:
Hi Randy, as I stated on the thread I opened on ardupilot.com (by your request), this was not a mechanical issue. what you highlighted as a mehcanical issue was the actual copter impacting my truck and suddently flipping to the ground.....that was the not the issue here.   take a look at attached please...thanks

[image]

govsux, 2015-12-09:
@imrj  So your saying with a very rapid loss of decent gps it might be better to disregard gps and even though it might drift it's better than shooting off in any direction?? I can see your point there. Maybe it should just land without gps if it'a within 5 meters without gps?

R-Lefebvre, 2015-12-09:
There are a lot of comments here which do not accurately describe the EKF and Inertial Navigation operation.  You cannot simply "stop using GPS" and expect to maintain position.  Using inertial navigation without GPS sends the copter randomly wandering off very quickly.

If you are in an environment without lots of clear space (at least 100 foot radius around the vehicle), you should be landing in Alt Hold or Stabilize mode.  These allow pilot control over position, and avoid the risks of GPS glitches when close to the ground.

imrj, 2015-12-09:
but Robert this is where I am confused...because there is already a LAND option for an EKF failsafe scenario...so how does that work (or does it not?)....are you saying that in that case it will land but not maintain position and drift off while landing then?

imrj, 2015-12-09:
I see now, thanks Robert. I guess the takeaway here is there is no way to have the copter lock into a position w/o some external sensor, the INS/EKF software alone isnt able to do that.

R-Lefebvre, 2015-12-09:
That is correct.  For manual landings, the pilots eyes are the external sensors.  But without something, using just the accelerometers, the copter will drift off quite quickly.