ArduPilot/ardupilot #3322: Cannot simulate gimbal operation

hamishwillee, 2015-12-10:
The correct way to simulate a gimbal is to pass the -M option to sim_vehicle.sh. Unfortunately this results in MAVProxy not being able to connect to SITL.

This was tested on Windows with latest master build of Copter.

peterbarker, 2015-12-10:
Seems to work OK under Linux, master Copter build.

Naterater, 2019-01-07:
@hamishwillee, is gimbal operation simulation still an issue?

hamishwillee, 2019-01-07:
Not for me - that's 2 years old and came up while documenting gimbal behaviour. Given that I won't have time to retest, I will  close this. Users can reopen or create a new one if they discover it is still a problem.

closed 2019-01-07 by hamishwillee

peterbarker, 2019-01-07:
https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/arducopter.py#L2059 <- that's the autotest for the simulated gimbal.