ArduPilot/ardupilot #3327: AP3.4 No logs on SITL

RoverHunter, 2015-12-11:
Using the last version of autopilot with SITL, no logs are saved under /ardupilot/audiplane/logs.
The file is actually created but no data is being recorded. File size is always 0KB.

On the DFMessageWriter.cpp file there the function void DFMessageWriter_DFLogStart::process(). On the case ls_blockwriter_stage_parms, I have noticed that the while(ap) cycle only runs 255 times, after that the next parameter is unreachable. Could this be due a 8 bit variable?

peterbarker, 2016-01-05:
@RoverHunter Apologies for the slow response.

I am unable to replicate this problem.  Log files are created, and they have > 255 parameters in them. 

Is this still happening for you?

WickedShell, 2016-04-18:
Closing as master does create complete logs successfully, and no further reproduction information has been provided (Or communication either way).

closed 2016-04-18 by WickedShell