ArduPilot/ardupilot #3328: Feature Request: Heading control in precision landing

derekxm, 2015-12-11:
Precision landing should be able to control the heading as well as the position of the copter. For example, if the copter is trying to landing on a battery swapping station, the heading of the copter is very important for battery swapping mechanism to engage correctly.

magicrub, 2015-12-11:
This is being addressed here: https://github.com/mavlink/mavlink/pull/432

rmackay9, 2015-12-12:
I think it'll take some changes on the copter side as well though so let's leave this one open I think.

Naterater, 2019-01-07:
Does precision landing control heading yet?

rmackay9, 2019-01-07:
No it doesn't I'm afraid

IamPete1, 2021-09-12:
@rishabsingh3003 is this issue still valid?

rishabsingh3003, 2021-09-12:
@IamPete1 yes this issue is still valid. At the moment we expect users to control yaw manually via RC. What does the user want? The heading should point towards the direction of where the copter is going (i.e point dront towards landing target)? I think this is a one-line trivial change and should be easy to do... otherwise, we can always control yaw via Lua/Mavlink scripts..