ArduPilot/ardupilot #3333: Calibration messages come through as high severity

DonLakeFlyer, 2015-12-12:
For example "Place vehicle on LEFT side and press any key" come through at high error severities instead of info level severities. This cause them to pop up as errors in QGroundControl.

peterbarker, 2017-12-19:
Challenge with this issue is finding a level which makes everyone happy.

peterbarker, 2020-11-01:
Could you enumerate those adjustments?  A reference to the relevant QGC code would suffice.

DonLakeFlyer, 2020-11-02:
Yup, here you go: https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/APM/APMFirmwarePlugin.cc#L253

WickedShell, 2020-11-02:
You shouldn't be using this message set anyways, we support a real set of MAVLink commands for acceleration calibration, which are much more robust, as we will resend them periodically in case the link drops anything. We use this command: https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L154

DonLakeFlyer, 2020-11-02:
I know. I still haven't had time to move to that. That said, even if I did that change I'd still need these severities fixed since these messages are still sent and will popup in the users face as errors.

peterbarker, 2020-11-03:
On Mon, 2 Nov 2020, Don Gagne wrote:

> Yup, here you go: https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/APM/APMFirmwarePlugin.cc#L253

I contend these ought to be NOTICE rather than INFO:

       <entry value="5" name="MAV_SEVERITY_NOTICE">
         <description>An unusual event has occurred, though not an error 
condition. This should be investigated for the root cause.</description>
       </entry>
       <entry value="6" name="MAV_SEVERITY_INFO">
         <description>Normal operational messages. Useful for logging. No 
action is required for these messages.</description>
       </entry>

What are your thoughts on the interactions with 
https://github.com/ArduPilot/ardupilot/pull/15242 ?

After that PR a user might choose to squash messages with a 
lower-than-parameter-value severity - meaning dropping the severity in 
ArduPilot would stop the user being prompted....

@tatsuy

DonLakeFlyer, 2020-11-03:
> After that PR a user might choose to squash messages with a lower-than-parameter-value severity - meaning dropping the severity in ArduPilot would stop the user being prompted....

FYI: I'm switching QGC to the `MAV_CMD_ACCELCAL_VEHICLE_POS` for accel cal. That said I still use the text messages to determine success/failure.

DonLakeFlyer, 2020-11-03:
> That said I still use the text messages to determine success/failure. Which would be a problem if these messages are suddenly not sent.

Nope that is incorrect. Just crap leftover code. I think accelcal was the last text based cal code in QGC.

WickedShell, 2020-11-03:
Moving to no text based cal code is great!