ArduPilot/ardupilot #3337: AP 3.4: Pixhawk only accepts 12 channels from CPPM but 16 from SUMD via Spektrum satellite port

marcmerlin, 2015-12-14:
With openlrs, if I output CPPM, pixhawk only receives 12 of the 16 channels (as seen in chanxx_raw in status)
If I output SUMD and wire it to go into the satellite receiver port, I see all 16 channels from my openlrs receiver in chanxx_raw.

Only problem is that using SUMD prevents me from using RX/TX on my receiver (i.e. I can have SUMD or RX/TX, while I can have RX/TX and CPPM), so I'd like to go back to CPPM.

Can AP be improved to receive all 16 channels from a CPPM stream, and not just 12?

marcmerlin, 2015-12-14:
I've been reading the code
void RCInput::_process_ppmsum_pulse(uint16_t width_usec)
in ./libraries/AP_HAL_Linux/RCInput.cpp
The PPM-sum code looks generic enough, it'll read channels until it hits a pulse of 2700 (not possible as a servo reading), or 
ppm_state._channel_counter >= LINUX_RC_INPUT_NUM_CHANNELS
and that's
./libraries/AP_HAL_Linux/RCInput.h:#define LINUX_RC_INPUT_NUM_CHANNELS 16

So I find no obvious reason why it would be limited to 12 channels.

magicrub, 2015-12-14:
@Marc,
You're looking at  ./libraries/AP_HAL_Linux/RCInput.cpp so does that mean
you're running on linux? I thought you were on Pixhawk.

-TomP

marcmerlin, 2015-12-14:
@magicrub oops, you are correct, I was grepping in the wrong place.
./libraries/AP_HAL_PX4/RCInput.h:#define RC_INPUT_MAX_CHANNELS 18
Is PX4 using ./Tools/ArduPPM/ (even though it's not arduino based?)

./libraries/AP_HAL_PX4/RCInput.cpp doens't have anything relating to PPM.

marcmerlin, 2016-02-02:
@tridge this is the 12 channel limit for PPMSum I noticed (but that isn't there for SBUS or SUMD)

Ravid824, 2016-07-14:
I run PPM 12 Chanel TX/RX and I don't anything in Mission Planner above Ch8
I run PM:Copter V3.3.3 on Pixhawk and the latest mission Planner. 
I don't know what to expect, should I see all channels in Radio calibration screen? Should I see the added channels 9-12 in the Extended Tuning window?
Below is the receiver PPM out signal which respose well to any TX channel changes. Frame length (Frame start to next frame start) is about 32 Msec , while after channel #12 it shows about 16 Msec until the next frame starts. 
Can some one direct me on what need to be done please.

[image]

OXINARF, 2016-07-14:
@Ravid824 support questions should be posted at the forum (http://discuss.ardupilot.org), please stop spamming the issues list.

Ravid824, 2016-07-15:
I will do so, sorry for this.

marcmerlin, 2017-06-10:
Another person with the same problem:
http://discuss.ardupilot.org/t/how-many-channels-supported-on-ppm-rc-input/18133

marcmerlin, 2017-06-11:
https://www.rcgroups.com/forums/showpost.php?p=37662843&postcount=10861
claims that 16 channel CPPM works on Rover, but everyone agrees that no more than 12 channels work on AP.

peterbarker, 2024-07-31:
This decoder has been rewritten several times now, so whatever bug this was is no longer present.  We have more modern bugs now.

closed 2024-07-31 by peterbarker