ArduPilot/ardupilot #3338: Very strange issue with older WP coordinates spawning at freshly uploaded mission.

AndKe, 2015-12-14:
.BIN https://www.dropbox.com/s/zrlr48ez4wpzvrd/Scout%20-%20badWPflight.BIN?dl=0
and openoffice spreadsheet that shows how the plane boots with old waypoints, then multiple uploads, and yet it tries to fly to an WP1 that were overwritten many times..
https://www.dropbox.com/s/3o5kg0uexjtiq47/Scout%20-%20badWPflight.ods?dl=0

Also: no amount of reading back the mission, got a bad waypoint down to GCS, every readback shows correct mission, but execution shows WP1 with old coordinates..

magicrub, 2015-12-14:
@AndKe I see this in gitter from you:
"A very strange, dangerous (I don't dare say bug) - behaviour. I sit on a ArduPlane .BIN from Pixhawk, the previous mission flown had 55 items (CTot) and was in another part of the country. The log proves that a new mission is uploaded, only in the northern part, 33items. when switched to AUTO, plane flew in wrong direction,(switched to manual/RTL plane navigated fine to RTL/circled fine) we uploaded/downloaded mission several times, NO bad waypoints. but the log proves that WP1 had bad coordinates, (did not knew it at that time) - deleted WP1 , - then log shows the new WP#1 had bad coordinates (!) - finally , a wild guess was to send it to WP#2 worked fine... and it flew well. Got similar issue next time too. The user uses current MissionPlanner. - but no amonut of mission write/readback showed these roque waypoints.. and I have no clue why the "first" always were bad, even after deleting the first WP..."

>>>"a new mission is uploaded, only in the northern part, 33items .... but the log proves that WP1 had bad coordinates,"
So the root of the problem is a partial mission write is corrupting WP 1? How are you writing? APMP2? MP?

AndKe, 2015-12-14:
All the affected logs belong to a plane owned by another company, they have only been using MissionPlanner during the missions. 
I have no reason to think there was an partial mission upload ever, we have reproduced it midair, as the log shows, re-uploading (and reading back) whole mission. Also, the first two uploads are done on the ground.

AquilaUAS, 2016-05-08:
Hi Guys, just reading this and I experienced a similar issue last Thursday whilst doing mapping flights, the first 2 flights completed as normal however the third somehow managed to complete 121 waypoint out of a total 187 and then proceeded to head off in the direction of the previous mission. When I looked at the BIN file the waypoints were those of the previous mission. I'm left confused as to what happened during flight 3. here is the dropbox link to the BIN file and the 3rd mission waypoint file. Thanks if you can offer any explanation.

               https://www.dropbox.com/sh/fm5wozoec9u2dz4/AAAADlixI6Pcsizno5SWkBDva?dl=0

David White

mboland, 2016-05-08:
When we are mapping we always do a read after a write of the waypoints, to make sure there was no corruption and all were written OK, as we are often mapping in sensitive areas.
Do you guys do a read after the write?
I would be concerned if you did and there was still WP corruption during the mission.

AquilaUAS, 2016-05-08:
To be honest a read after upload has never been done, though from now on it will be. I just find it concerning that just over half of the mission completed before the previous one apparently kicked in.

rmackay9, 2016-05-09:
It makes sense that we should protect against half written missions in the autopilot code and/or in the ground station.  It's not immediately clear to me how we should do that.  I suspect that the ground station, when uploading an entire mission, should first clear the existing mission and check that the mission has been cleared somehow (i.e ack/nak from the clear command).  .. so at the risk of passing the buck, I suspect it's more of a ground station issue but there could be some loopholes in the ardupilot code as well.

AndKe, 2016-05-09:
ApmPlanner2 does already that, writes a mission, then reads it back,
visualizing it while doing so.
The acks and ability to read back should be enough if GCS is well made and
cares to check.

WickedShell, 2016-05-09:
A implicit clear every upload can be extremely poor if the aircraft hits a waypoint that it was flying it, but hasn't received the next one yet (will cause a mode change, which the user wasn't expecting). A better solution for that part is atomic uploads which I will be looking into over the next couple of weeks

magicrub, 2016-05-09:
I often upload an auto mission while flying auto because all I'm doing is
changing a param of a single mission cmd. I would really hate it if it
wiped before it uploaded, it would trigger an RTL and lose the mission
index.

AndKe, 2016-05-09:
Sometimes I too change/upload a new part of mission midair, but always
switch to loiter, upload, verify -if mission is supposed to resume from a
certain WP, I add a jump command in the beginning.

AquilaUAS, 2016-05-09:
Looking at the messages played back from the tlog, close to the point where the plane deviated from the mission I see a double "Execute nav command ID#16" . There are no other double command in the log. In this instance the mission upload was commanded whilst plane was stationary and unarmed on the ground. I'm beginning to think its a MP issue and not FC. I had a couple of failed mission uploads on an earlier mission that day (unknown reason). Its making me now think to do a shut down on the GCS after each mission even though I like to ensure I am getting decent overlap of missions, I may have to relook at that workflow. I am due out to do another 5-6 flight later this week so will adopt a new strategy then. Below is the a snapshot of the commands as reported from the tlog:

Executing nav command ID #16
Reached Waypoint #121 dist 18m
Executing nav command ID #16
Executing nav command ID #16
Reached Waypoint #119 dist 19m
Executing nav command ID #16
Passed Waypoint #118 dist 2m
Executing nav command ID #16
Passed Waypoint #117 dist 21m
Executing nav command ID #16
Passed Waypoint #116 dist 21m
Executing nav command ID #16
Reached Waypoint #115 dist 19m
Executing nav command ID #16

 I have put the tlog in the dropbox folder . The 2 logs to look at are 2016-05-05 14-39-35 & BIN442. The mission previous to this (the one that the plane reverted to after WP121) are 2016-05-05-13-03-31 & BIN441 https://www.dropbox.com/sh/fm5wozoec9u2dz4/AAAADlixI6Pcsizno5SWkBDva?dl=0

magicrub, 2016-05-09:
Ok, I'll close this. @AquilaUAS can you post an issue about this on Mission Planner's issue page?

closed 2016-05-09 by magicrub