ArduPilot/ardupilot #3344: Add Precision Loiter to Copter

Chambana, 2015-12-17:
Copter would benefit from a Precision Loiter capability in Loiter mode, similar to the Precision Landing capability in Landing mode.  Perhaps Precision Loiter could be implemented with a LOITER_TARGET message, similar to how Precision Landing uses LANDING_TARGET messages?  

A "Precision Loiter" capability can be demonstrated in 3.4 master by simply setting the parameter LAND_SPEED to 0 while in Land Mode and sending a LANDING_TARGET message.  The copter will loiter over the ground position defined in the LANDING_TARGET message.

rmackay9, 2015-12-17:
indeed a few people have asked for this.  Txs for the request!

R-Lefebvre, 2015-12-17:
Interesting.  So does the loiter end up being more precise than it would be just on GPS alone?  ie: less movement?  Or is it more about loitering over an *exact* location, but possibly with some position dithering?

squilter, 2015-12-17:
@R-Lefebvre It could also be used for hovering over a moving object.

Maybe we could just rename the PRECISION_LAND message to PRECISION_SOMETHING and use it for both.

govsux, 2015-12-17:
Is landing more precise in holding it's location than loiter / pos hold? The only thing the FC can use is gps unless it has optical flow right? Or maybe that's what's being suggested?

R-Lefebvre, 2015-12-17:
I think part of the problem here is that using the terminology "Precision-Whatever" might lead to the assumption that this control method is more precise than others.  I'm not sure it is.

What we are really talking about here, is controlling position relative to an optical target on the ground.   The relative precision of that operation is not certain.

Chambana, 2015-12-22:
Yes, I'm looking to control position relative to an optical target on the ground.

Basically, my companion computer is already calculating my x-offset and y_offset for the LANDING_TARGET messages that get sent to flight controller at ~10Hz. It works really well and it would be nice if I could just send those same offsets in some kind of message that updates the Loiter target (for a static or moving target and in loiter mode not land mode). 

Yes, I am using Optical Flow not GPS.

rmackay9, 2017-01-03:
Peter's implemented this so it will go out with AC3.5 which will start beta testing in January.

closed 2017-01-03 by rmackay9