ArduPilot/ardupilot #3345: AP Feature Request - RTH Stepped Altitude

SlipstreamFPV, 2015-12-17:
Would love to see altitude steps (or just one step) added to RTH mode in Arduplane.

e.g.

Alt Hold RTH when distance from home is less than "...." m = 80m
Alt Hold RTH when distance from home is greater than "...." m = 150m

SlipstreamFPV, 2016-02-14:
Any response to this? How about an even easier setup: A setting where ALT hold RTL is the minimum alt for RTL and if your current alt is above that then the plane will return at your current altitude? Using the current -1 setting is too risky incase the plane is below the usual rtl height.

magicrub, 2016-02-14:
For plane it sounds like a good enough idea having an RTL_ALT param. How
would that work in with DO_LAND_START?

SlipstreamFPV, 2016-02-15:
Maybe a waypoint set directly above the final landing waypoint? Once that waypoint is reached then circle down to the normal RTL alt and then kick back into the set landing pattern. Should work the same for planes without landing too. Return to a waypoint above home at the current alt then circle down until the normal RTL alt.

magicrub, 2016-02-15:
great, so we can do that already! :)

kudykam, 2016-06-29:
I thing even better would be when RTH is triggered, then head home and descent to RTL altitude gradually. Which will speed up returning process... Current status is very stupid, travel at same altitude and descent over RTH location.

Naterater, 2019-01-04:
I think this can be accomplished with a "landing sequence" that is triggered by RTL_AUTOLAND. Instead of actually having a land, you just have a waypoint (or a few) followed by a RTL (no land involved). Can this be closed?

SlipstreamFPV, 2019-01-04:
That would require setting up waypoints for each flight which is frustrating.

What about just a two step RTL. If distance from home < x then rtl alt = a. If distance from home is > x then rtl alt = b.