ArduPilot/ardupilot #3346: AP_AHRS_Nav_EKF: Can't set ekf type 2

carolinux, 2015-12-17:
Hello,
I am trying to customize Replay to replay some custom logs. I found this bit in the code:
https://github.com/diydrones/ardupilot/blob/1e8c3910241a03975699a3891f3efbc954b9193b/libraries/AP_AHRS/AP_AHRS_NavEKF.cpp#L754

and I wonder if it's a bug or it is intended that one cannot set the ekf type to 2. (Even though 2 is used as a value in switch statements later or mentioned in the comment here https://github.com/diydrones/ardupilot/blob/1e8c3910241a03975699a3891f3efbc954b9193b/libraries/AP_AHRS/AP_AHRS_NavEKF.cpp#L598). I don't see a way that ekf type can be set to 2 from master.

peterbarker, 2016-01-05:
@carolinux That code only determines whether *any* EKF is used.

EKF type is set by the AHRS_EKF_TYPE parameter:

STABILIZE> 
STABILIZE> param help EKF_ENABLE
STABILIZE> EKF_ENABLE: Enable EKF1

This enables EKF1 to be disabled when using alternative algorithms. When disabling it, the alternate EKF2 estimator must be enabled by setting EK2_ENABLED = 1 and flight control algorithms must be set to use the alternative estimator by setting AHRS_EKF_TYPE = 2.

Values: 
	0 : Disabled
	 1 : Enabled
param help AHRS_EKF_TYPE
STABILIZE> AHRS_EKF_TYPE: Use NavEKF Kalman filter for attitude and position estimation

This controls whether the NavEKF Kalman filter is used for attitude and position estimation and whether fallback to the DCM algorithm is allowed. Note that on copters "disabled" is not available, and will be the same as "enabled - no fallback"

Values: 
	0 : Disabled
	1 : Enabled
	2 : Enable EKF2

WickedShell, 2018-08-26:
The code in question no longer exists, so closing out the issue.

closed 2018-08-26 by WickedShell