ArduPilot/ardupilot #3348: Copter: Add precision landing status to telemetry

derekxm, 2015-12-17:
ArduCopter currently logs precision landing to DataFlash log. But it will be very useful to also see precision landing status in real time telemetry through Mavlink.

squilter, 2015-12-17:
What information are you specifically hoping to see?

derekxm, 2015-12-17:
uint64_t time_us;
    uint8_t healthy;
    float bf_angle_x;
    float bf_angle_y;
    float ef_angle_x;
    float ef_angle_y;
    float pos_x;
    float pos_y;

Following Log implementation, these will be nice.

squilter, 2015-12-17:
What is the specific use case where you want to see this info?  This is very similar to the information that is sent to ArduCopter with the [LANDING_TARGET](http://mavlink.org/messages/common#LANDING_TARGET) message.  If you are developing something yourself that uses precision land, then you already have this info.

derekxm, 2015-12-17:
It will mainly be used to debug and analyze precision landing with IR-Lock sensor in real time. Since LANDING_TARGET message is only used with AC_PrecLand_Companion, our computer do not know the state of precision landing with IR-LOCK. Therefore, a reverse of LANDING_TARGET message that communicates to a companion computer is desired.

Chambana, 2016-01-03:
Agreed, this would be very useful for debugging/analyzing precision landing solutions.  

Also, providing the precision landing status over telemetry would help the GCS and/or Companion Computer diagnose, post-landing, how close the copter was to landing on the IR-Lock beacon (or whether it missed the beacon entirely).  

@derekxm IR-Lock has a Python module that your companion computer can use to access the IR-Lock detection data directly.  This should give your companion computer the same data the pixhawk gets from the sensor.

pryre, 2022-01-12:
It would be very helpful from a monitoring point of view to simply know the status of a precision landing system from the GCS. I would propose to use the new `MAV_SYS_STATUS_SENSOR_EXTENDED` in MAVLINK to pass this information down (if system is enabled and/or if it is currently active).