ArduPilot/ardupilot #3354: AP: RTL should have a minimum ground speed setting

marcmerlin, 2015-12-19:
I typically set my trim airspeed for automatic flying to something not too high since adding more power just burns exponentially more batteries without giving a lot of extra speed, and I don't want circle or loiter to use 20A when it could be using 5A.

However, if I lose RC control, RTL kicks in and flies back. Sadly I have seen this happen on video where the airspeed was minimal (enough to fly ok, as I requested), but making almost no progress on returning because of a headwind.
If I have RC control, I can tell AP to go faster by sending more throttle. That's good.
If I don't have RC control, AP will stay at low throttle and fly with almost no ground speed when there is a stiff headwind.
I would love for a minimum groundspeed setting where AP will in all auto modes use enough throttle to hit that groundspeed, or hit max allowable throttle for auto modes.

Thanks.

magicrub, 2015-12-19:
Implemented as Min_gndspd_cm

http://plane.ardupilot.com/wiki/arduplane-parameters/#minimum_ground_speed_arduplanemin_gndspd_cm

marcmerlin, 2015-12-19:
Argh, you're faster than me, I was just going to update the bug to be more clear. I think what I was hoping for is unclear. If I loiter, I don't want to spend a lot of motor power to sustain MIN_GNDSPD_CM but if I RTL, I'd want that.
But the problem may just be that MIN_GNDSPD_CM does not do what I think it does and it may not apply for LOITER. 
If I'm stuck having CRUISE go at MIN_GNDSPD_CM (not always what I'd want if I'm doing a slow overfly to video something), to get MIN_GNDSPD_CM for RTL, it's a worthwhile tradeof for now.
My use case may be borderline, so it's fine to leave this closed.

magicrub, 2015-12-19:
I don't think I understand what you are among for. This param will push the throttle up if ground speed is below that value. Works in any mofe where the throttle is controlled by the autopilot.  A value like 500 works great. Can you give an example scenario to help clarify?

marcmerlin, 2015-12-19:
If I'm in FBWB or CRUISE, I'm in control of where the aircraft is going, and I may be happy to hover over the ground at 5kph ground speed to take aerial shots. 
If the airplane is in AUTO, but specifically RTL, it is trying to get somewhere, hopefully in a battery efficient way.
In the first case, I'm happy with my ground speed potentially being very small.
In the 2nd case, I'm not :)

Does that make sense?
Now, this is kind of a far out wishlist, I can just set MIN_GNDSPD_CM and switch from CRUISE to FBWA if I want to hover at lwo groundspeed, so it's still ok to leave this RFE closed, but hopefully it makes more sense now how I was looking at both cases as being separate.

R-Lefebvre, 2015-12-19:
This is very similar to a request in Arducopter.  The issue is that we only have one speed setting.  Say you are mapping at a target speed of 10 m/s, and have flown quite far away from the operator.  Then you have a battery or radio failsafe.  The copter might not make it back in time.  What is needed is different speed settings.  1 for the waypoint flying.  Then a higher speed "transit" speed to get the copter back home before the battery runs out.

magicrub, 2015-12-19:
I've re-opened the issue and changed the title to "Create dual min gnd speed: RTL (new) and everything else (MIN_GNDSPD_CM)"

marcmerlin, 2015-12-19:
Actually I'd separate RTL+AUTO (I'm not flying, and want to aircraft to get there, I may not have video to check, so getting there quickly against a headwind, is a must)
and
CRUISE+FBW, where I'm flying and potentially flying via video. Here the ground speed is not as relevant since the aircraft does not know that it has to get anywhere, or maybe hovering in place is actually desired.

gelazanskas, 2015-12-26:
I would go for two separate MIN_GNDSPD_CM's - one for RTL and the other for everything else.

marcmerlin, 2015-12-26:
@gelazanskas you likely want one for auto modes, computer is flying, just get there, vs manual flying, I decide where I'm going and how fast, and minimal ground speed doesn't matter potentially since I'm in control and I may want to fly slowly or fast.

magicrub, 2015-12-26:
Auto and RTL are both auto modes.

marcmerlin, 2015-12-26:
@magicrub yes, that's what I was trying to tell @gelazanskas when he said "RTL" vs "everythiing else" :)

magicrub, 2015-12-26:
But we're talking about RTL and everything else, right?

magicrub, 2015-12-26:
And to be clear, we're talking about an enforced min ground speed, not airspeed.

marcmerlin, 2015-12-26:
We're talking min ground speed for any mode primary flown by the flight controller, which is at least AUTO and RTL
Any FC assisted or manual flight mode (MANU, FBW*, TRAINING, etc), ground speed can be 0 or even be negative,that's the pilot's problem :)

magicrub, 2015-12-26:
Awwww I think I get it. The real problem is that it is currently enforced
in fbw and you'd rather it not?

marcmerlin, 2015-12-26:
Correct, FBWB and CRUISE. I mostly use CRUISE for flying while taking shots, and I wouldn't want it to fly at high ground speed, just because it makes sense to do so for RTL.

marcmerlin, 2016-02-02:
@tridge, as discussed, any AUTO navigation mode (including RTL while getting back, or GUIDED getting back) would use a MIN_GNDSPD_CM and it would be ignored otherwise because pilot is deciding where to go, how fast, and whether almost hovering over a point in headwind is actually desired.

WickedShell, 2017-05-29:
Haven't looked at this one for a bit, but I'm of the opinion that at a minimum cruise should also account for MIN_GNDSPD_CM as it's a mode that is used to travel a distance over the ground/return the aircraft somewhere.

Naterater, 2019-01-04:
Can't we make everyone happy by providing a bitmask for the events where Minimum Groundspeed is enforced?

MIN_GNDSPD_BTMSK - Modes where Minimum Groundspeed is enforced.

Options: 
AUTO
CRUISE
FBWB
GUIDED_TRAVEL
GUIDED_LOITER
LOITER
RTL_TRAVEL
RTL_LOITER

Naterater, 2019-01-04:
As a semi-related comment, it is very frustrating during mapping missions when turning into the wind causes a huge increase in airspeed (and resultant turn radius). I actually only want this to be enforced when the distance to the waypoint is still > ~200m. When I'm closer to waypoints, I like my turns to be sharper (at a lower airspeed). Maybe a MIN_GNDSPD_DIST could be enforced as well to only work when distance to the waypoint is > the value set.