ArduPilot/ardupilot #3356: Allow simultaneous RC and Joystick control

heinduplessis, 2015-12-20:
I see that if one uses a joystick on MP, RC control no longer works until the feature is disabled via the drop downs in the joystick setup screen in MP. This is rather dangerous since the telemetry can be lost and then one can no longer control the airplane at all.

Is it possible to allow simultaneous RC and Joystick control?

R-Lefebvre, 2015-12-20:
How would the logic work to allow *simultaneous* control via both inputs?  If the roll input on one is 0 degrees, and the roll input on the other is 15 degrees, which do we act on?  One possible answer would be "the one that is not zero".  But what if the second system has failed and is stuck at 15 degrees, and we are trying to fly on the primary with a zero roll input?

heinduplessis, 2015-12-20:
RC priority?

R-Lefebvre, 2015-12-20:
As in, a new parameter?

mboland, 2015-12-20:
Hightec RC Transmitters work this way in buddy box/trainer mode and although I was sceptical at first as an RC Instructor I found it very helpful.
To answer your query Robert, you would be holding 15 deg of opposite roll on the primary system if the secondary was stuck at 15 deg.
But I see your point and maybe it would be good to be able to switch the secondary control on or off the same way we do for RC controllers when training, the primary having a switch to initiate or deactivate it.

R-Lefebvre, 2015-12-20:
Yes, that's what I've been looking to add.  You would define which is primary control input (PWM) and which is secondary (MAVlink) or vice-versa.  Then you have to explicitly hand over control.  It might be a bit inconvenient, but would solve a lot of problems.

magicrub, 2015-12-20:
The way I've seen these dual-control systems work is to take the command
from whaichever one most recently changed more than a given deadband
percent. This is to not trigger on noise. 1% may be enough, maybe 3 or 5%.

R-Lefebvre, 2015-12-20:
I guess that's another way.  Still not as robust as explicit switching.

This ties in with the Input Manager class I'm working on.  I planned to discuss it on the next dev call.  Unfortunately, it's a bit of a 3rd Rail.  Every time we talk about this stuff, we end up in an impassible disagreement.

WickedShell, 2015-12-20:
I think it'd make sense to allow a channel to function as an overide. IE in
our plane systems we operate with a multiplexer that allows the RC control
to completely override anything the autopilot is trying to do. It's
explicitly controlled by a switch on the transmitter and if you want the
autopilot to be able to fly then you transfer the control with the RC.
(Thus RC input is always the master) We've found this to be quite reliable
and clear.

heinduplessis, 2015-12-20:
@R-Lefebvre  I'm sure we can all agree that RC should have priority. If all else fails, one reaches for the transmitter!

So, perhaps the code can be modified to ignore Mavlink movements if there are RC movements. Kind of what's happening in Cruise mode - APM detects RC input and acts as FBW until no inputs are received, then it would act as Auto with WP 1 km ahead.

Of course, if your trims are set outside of APM, APM will think you're continuously using the RC, so Mavlink joystick movements will be ignored, but that's how AutoTune  and Cruise also works.

@WickedShell an override channel might be a problem for some (I'm constantly out of channels) and perhaps will need too much of a code change. I could be wrong. 

Option 3 - Perhaps a completely new mode? A "Cruise" mode that only listens to Mavlink?

rmackay9, 2015-12-21:
I think if you're using the joystick with mission planner and just pull the USB connector out of the mission planner it will return control to the transmitter.  Not saying that's the most elegant way to do it but it should work.

severinleuenberger, 2015-12-21:
Pulling the Joystick-USB-Cable is at least for our Groundstation not an option, because the USB-Plug is for security-reasons directly sealed against the Groundstation-PC-Housing (with PlastiDip). So that it doesnt unplug accidentally.... Therefore - this solution would for our case not really help. And - I would love to have a solution where the primary steering is with RC-Radio and the second comes from the groundstaion joystick. And - yes - it would be great, if the RC-Radio steering input does disable the joystick...

heinduplessis, 2016-01-10:
Agreed with @severinleuenberger !

Eyal-Erez, 2017-09-18:
Hi All,
So where does it stand now? is there an option/cmd to switch between RC/Joystick??

geofrancis, 2017-09-18:
+1 for this. It's the sole reason I have never used a joystick since there was no fast way of retaking control with the handset.

rcywongaa, 2017-10-27:
+1, very important when using a companion computer that controls drone movement.  If the computer crashes, user needs to be able to retake control with their RC controller.

Currently, the best way to handle the situation where a companion computer crashes after issuing an RC override is to immediately switch off the user handset to force a failsafe... 

It also seems somewhat inconsistent that fail safe isn't triggered automatically even though the user handset is useless in this situation

magicrub, 2017-10-28:
@rufus your companion computer should not be sending RC commands for flight
for exactly that reason (and delay and many other reasons). Please use
guided mode.

cheintz, 2017-11-07:
+1, because there is no 'guided mode' for planes. I crashed an airframe into the ground at 60 mph when an odd set of circumstances caused the plane to be in manual mode with locked control surfaces.

Do the RC overrides get cancelled when the actual RC transmitter goes to failsafe?

magicrub, 2017-11-13:
yes, there IS a guided mode.

cheintz, 2017-11-13:
There is a guided mode, but as far as I know, it doesn't provide the
velocity/attitude command interface that Copter does. Thus, RC commands are
needed even in that mode to achieve any control more detailed than "fly to
here and circle".

magicrub, 2017-11-14:
> it doesn't provide the velocity/attitude command interface

Have you tried? It's there...

MAVLINK_MSG_ID_SET_ATTITUDE_TARGET is implemented [here](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/GCS_Mavlink.cpp#L1646). However, I advise against using it unless you are doing something VERY special [like this](https://www.youtube.com/watch?v=RSn62JKIRks). Are you running an external flight controller? The usual flight controller is doing all the low-level tight-loop stuff and the external system is suppose to provide higher-level navigation. that's what GUIDED mode is. How are you using?

MAV_CMD_DO_CHANGE_SPEED is implemented [here](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/GCS_Mavlink.cpp#L1046).
 

>Thus, RC commands are needed even in that mode to achieve any control more detailed than "fly to here and circle".

Or... give it a waypoint far away and keep updating it at some slower interval, like 1s (or faster if you wish).. That lets the flight controller do the closed-loop stuff that it's good at and the slower unlimited memory device can point it around on a more comp0licated trajectory mision

WickedShell, 2018-07-17:
Alright I want to summarize some changes, and then see what people are still interested in chasing from this issue.

The most significant improvement to have happened recently on this front will be shipping with Copter 3.6 and Plane 3.9. There is now a `RC_OVERRIDE_TIME` parameter that controls how long to keep using an override value if it hasn't been updated. So if this is set to 3 for example, if the GCS has not gotten any commands to the vehicle within the last 3 seconds the control will fall back to the RC transmitter. This seems to mostly address the original issue.

Along the way two other topics were brought up:
1. @mboland mentioned this HiTec mode where we simply leveraged the inputs from the overrides and the RC transmitter against each other. I'm not a huge fan of this, as you still end up with a dead lock (+45 from one, -45 from the other), it could interact very weirdly with a RC transmitter that is failsafeing (particularly since this is tracked/managed up in the vehicle code rather then the RC input code).
2. Mapping of a channel on the RC transmitter to disable overrides. I'm in favour of this solution and think it holds real value. For easy/consistent implementation it needs to wait on a bit of other clean up work that is ongoing at this time, but it could then be done fairly easily.

Are there any complaints/issues with changing this issue to just be a switch to disable overrides?

lebarsfa, 2021-05-14:
Hello,

Just to get cross-reference between related issues/comments : I think topic 2. of https://github.com/ArduPilot/ardupilot/issues/3356#issuecomment-405733149 is linked to https://github.com/ArduPilot/ardupilot/issues/7045#issuecomment-841299809.

peterbarker, 2021-05-17:
We have RC-override timeouts *and* you can have a transmitter switch to disable overrides.  This satisfies the original problem description.

Closing this now.

closed 2021-05-17 by peterbarker