ArduPilot/ardupilot #3370: logic for gps-glitch and -fault using two gps-modules on a pixhawk with APM plane

severinleuenberger, 2015-12-22:
Diverting from a healthy working GPS,  there can be 3 status, that are problematic: 
- GPS-glitches
- GPS-freeze (constantly the same freezed and wrong signal)
- GPS-failure (no signal)

While using 2 instead of one GPS-module, I wonder if the APM logic deciding to use the correct GPS module does really choose the right and best working GPS system, and does warn with correct and helpful warn-messages over MavLink accorgingly about unhealthy GPS-stautses in one of the modules.

I like to discuss here the existing logic and start the logic to improve the logic, so we have a somehow "bulletproof" GPS-System. We should also consider using  NAVSTAR GPS and GLONASS as well, maybe in combined modules or separated ones.

WickedShell, 2015-12-23:
The current logic is pretty primitive, the first point is that whichever GPS is reporting the higher status (DGPS > 3D GPS > 2DGPS etc) will be used first, if the status level is the same it will swap to whichever satellite has more satellites. There is a timer on how often the GPS that is being used as primary can be swapped, if the GPS hasn't been swapped for the last 5 seconds and there is a reciever that has 2 more satellites then that will be used, if there hasn't been a swap for 20 seconds and there is a reciever that has 1 more satellite it will swap which GPS is being used.This is all available here: https://github.com/diydrones/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.cpp#L390

There is certainly room for improvement in the GPS swapping algorithms, but most efforts in that direction haven't been able to handle the variety of hardware combinations available. For example using a GPS that combines GPS and GLONASS will give a higher satellite fix (and thus priority) then a GPS that does just GPS, even though the GPS only receiver might have a better fix.

When swapping GPS units there is a noticeable difference in position, which is why we try to limit how often we swap, (primarily noticeable on copter when hovering). The MAVLink messages about which GPS is the primary is a big weakness and there is already an issue open about that see #2572

WickedShell, 2017-05-29:
There is a blending mode present in master/copter 3.5rc7, it presents a much more advanced solution for GPS selection. Blending does take into account how old the most recent data is. Closing this for now until there is a specific complaint with the blending mode, or a different proposal for alternate algorithms.

closed 2017-05-29 by WickedShell