ArduPilot/ardupilot #3374: Gripper implementation does not match MAVLink definition

hamishwillee, 2015-12-23:
The MAVLink definition of [MAV_CMD_DO_GRIPPER](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L26) states that gripper number is from 1 to max grippers.

According to @rmackay9 we have implemented it as 0 indexed - so we should either match the protocol or get the protocol changed.

squilter, 2015-12-29:
I think changing the protocol would be the cleanest way to do this.

OXINARF, 2017-06-02:
I'm not sure where this issue came from since by looking at history the gripper number was always (and still is) ignored. As we only support one gripper at this time, this isn't very important - if we start supporting more we should verify then that the protocol is followed or changed, but it isn't a current issue.

closed 2017-06-02 by OXINARF