ArduPilot/ardupilot #3384: Invert Flying

Pomaroli, 2015-12-26:
Is it be possible to add a new 3D functionality which could make invert flying possible by turning motor direction?
(mechanical pre condtitions are 3D Props and ESC's whit invert SimonK firmware)

Pomaroli, 2016-04-01:
Hi Guys, how is it going with this request ?
i've build a smaller quad with NAZE32 controller and Cleanflight Firmware which already has the 3D Option. The ESC are DYS with BLHeli Firmware which can setup for Bi-Directional use.
Basically the 3D option in Cleanflight is very rudimentary,  the throttle middle position  is equal the idle position (DZ +-50 configurable), A switch is configured to Arm/Disarm the Copter. If the Copter is Armed and the Throttle Stick is pushed upwards the Motors will Spin "Normal"  in case the Throttle Stick is pulled downwards the Motors will spin reverse.

IamPete1, 2021-07-25:
You can do this with the 6DoF scripting motors backend. Happy to help with the required lua script if anyone want to try it, in the mean time i'm going to close this.

closed 2021-07-25 by IamPete1

Pomaroli, 2021-10-30:
Hi Pete as i undestandig right there is a free confirgurable FRAME_CLASS: Frame Class 6Dof which i can use for this purpose, Well i would need help to step in to, is there any good learn video or howto ?

IamPete1, 2021-10-30:
@Pomaroli my confrence talk is probably a good place to start, https://youtu.be/QUDhnYvH66k

Maybe start a thread over on the forum and I can help you get setup if you get stuck.

Pomaroli, 2021-10-31:
Pete thanks to link me the conf. video, very exciting to listen to. Im not sure if we have the same pov. There is a really deep technical pov where the question is what should be the goal of a 6dof and how to deal with. In my pov we are in a scenario where we are in acro mode, the behaviour of the uav should be controlled by the pilot, the uav high and position is relativ. I wrote in the opening the exmple with the throttle push it goes up and when pull the motor direction invert. But this is spoken really simple in real live with this behaviour it is possible to start doing really nice manouvers like rolling the uav sitting on the same spot or flying an altitude 8 curve. I do this with my 450 quad on naze32 imagine this figure on a blue sky thats some what of cool. I will need to do some videos for better understanding. Are we still on a 6dof approach with this scenario? If you can confirm i try.