ArduPilot/ardupilot #3389: Failovers don't fail if USB joystick disconnected

hamishwillee, 2015-12-28:
No failover to transmitter if the usb joystick is disconnected from the usb port, or if I cut the telemetry link, I followed the [exact procedure from the page](http://copter.ardupilot.com/wiki/flying-with-a-joystickgamepad-instead-of-rc-controller/), this was tested 6 months ago, and again today after 2 updates of mission planner, on both a apm and pixhawk.

Posted as https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/255

robustini, 2017-08-20:
Same issue here with the latest code in the master.
And is not a Planner issue.
There must be a possibility to override the joystick by tx if the telemetry fail.

rmackay9, 2017-08-21:
We should clarify if this is Plane or Copter (or something else).  I'm pretty sure this works ok with Copter.

WickedShell, 2019-04-05:
This was fixed/improved awhile ago. The `RC_OVERRIDE_TIME` parameter specifies a time after which no new inputs have been received that will cause us to ditch your overrides. This defaults to 3 seconds. So 3 seconds after your GCS stops sending packets control will revert to your RC transmitter. (If your GCS continues to spam even without an input device then that is a GCS issue)

closed 2019-04-05 by WickedShell

rrr6399, 2019-04-05:
Has the GCS failsafe been tested with no R/C turned on? (i.e. controlling with a Joystick and telemetry fails, which triggers an RTL.)

WickedShell, 2019-04-05:
@rrr6399 I've tested that numerous times (in SITL) and it behaves as I expect it to. From a vehicle perspective once you're override ends there is no new RC input, which starts the timer on the RC failsafe.

rrr6399, 2019-04-05:
Thanks for the response @WickedShell . Did you happen to test what happens if there is no RC receiver or the receiver zeros out the inputs when the RC is turned off?

WickedShell, 2019-04-05:
I'd actually used the RC_OPTION to ignore any input from the receiver when I was doing the testing.

rmackay9, 2019-04-06:
Jaime tested with Rover a few months ago and we [found an issue with MP that's been fixed now](https://github.com/ArduPilot/MissionPlanner/issues/1972).

I'm pretty confident that the latest versions of Copter and Rover correctly trigger a failsafe when contact is lost with the transmitter whether that's a tranmitter sending rc-overrides via mavlink or a normal tx/rx setup.