ArduPilot/ardupilot #3393: GCS param writes occasionally timeout

magicrub, 2015-12-29:
GCS param writes occasionally timeout. Happens on multiple GCSes when writing bulk params. @peterbarker has duplicated on master.

peterbarker, 2015-12-30:
My duplication was traced to a bad telemetry radio.  Replacing the radio fixed the issue for me.

I am now unable to reproduce this issue on master, sorry.

Peter

magicrub, 2015-12-30:
<sadface>

magicrub, 2015-12-30:
bummer! Ok, I'll keep poking around. Thanks for your help!

imrj, 2015-12-30:
this has happened to me too, it does seem to be when more than one GCS connected, but not all the time either

magicrub, 2015-12-31:
Happens to me with a single companion computer connected. No "GCS" per se.

magicrub, 2016-02-21:
this was tracked down to our GCS behavior. We were writing 450 params to the pixhawks which is not a very good idea. We're now doing a read modify write operation and is much faster and stable.

closed 2016-02-21 by magicrub