ArduPilot/ardupilot #3400: Make check should always run the tests

lucasdemarchi, 2015-12-30:
If tests are already compiled, it isn't running them.

guludo, 2016-01-04:
@lucasdemarchi waf just runs the tests if it thinks they're necessary. So if you've run a test and you don't change the code (neither clean the build), then it won't run the test for that binary again.

That behavior is pretty solid in my opinion. If you want all tests to run anyway, you can use something like `waf check --alltests` or `make -f Makefile.waf check WAF_FLAGS='--alltests'`.

Thus, I don't think we make check should always run the tests by default.

lucasdemarchi, 2016-01-04:
@guludo it's not consistency with other build systems. We need consistency. Go to an autotools or cmake based build system and call "make check": it compiles (if needed) and runs all tests. Arguably the tests should only be build by `make check` as well.

guludo, 2016-01-05:
@lucasdemarchi 

> @guludo it's not consistent with other build systems. We need consistency. Go to an autotools or cmake based build system and call "make check": it compiles (if needed) and runs all tests.

I personally don't see much benefit of keeping it consistent with the other build systems since the way that is done by waf looks more efficient in my opinion. Why do we need consistency in this case? After all, we're using a *different* build system, with different approaches, that may or not be better than ones taken by other build systems. I think the key question here is: How do the benefits/disadvantages of keeping it consistent with other known build systems compare with the "waf way"?

> Arguably the tests should only be build by make check as well.

I remember when I was implementing that, my first approach was to add the test programs to the graph only when a `check` command was issued. But for some reason, which I don't quite remember, @cmarcelo pointed that we needed the graph "complete" for all builds. Could you help us here @cmarcelo? hehe

cmarcelo, 2016-01-05:
Note that having the full graph doesn't imply that you'll actually build all the targets. You probably can fix up the target list in build to pick the right targets. Just make sure when targets are explicitly set, they will be respected (i.e. you may want to change the behavior of `waf build` with no arguments, but if the target list is sent, you want to respect it).

The reason for keeping the full graph is that object files used multiple times have different extensions (there's a numeric id for those). Having partial graphs might affect these ids. There are ways to avoid constructing the full graph, but I'm not convinced we need to.

lucasdemarchi, 2016-01-05:
Then maybe add a `test` target to the make wrapper. The reason is the build-compile-test cycle of whom is writing a test: imagine you have 2 failing tests.

 - you run everything by calling `make check`
 - you change something to fix one of the tests and run `make check` again
 - there will be no report of the other still failing test since it won't be executed

If you add a `test` target either to waf or to the make wrapper then I'm happy :-). It could plain simply call the `check --alltests`, which is too big to keep typing.

cmarcelo, 2016-01-05:
> If you add a test target either to waf or to the make wrapper then I'm happy :-). It could plain simply call the check --alltests, which is too big to keep typing.

Other way to think is that we may got the default case wrong. Let `waf check` compile _and_ run the tests, then add an option to "do not run tests please".

guludo, 2016-01-05:
@lucasdemarchi 
> there will be no report of the other still failing test since it won't be executed

You're right and you convinced me :-) I think we could make `waf check` compile and run all tests for now.

However, I think the ideal would be to only run relevant tests, which would be:
 - tests whose binaries or dependencies have changed (given for free by waf)
 - tests that fail

We could do that in a later work.

@cmarcelo 
> Other way to think is that we may got the default case wrong. Let waf check compile and run the tests, then add an option to "do not run tests please".

waf already provides a command line option for that: `--notests` :-)

guludo, 2016-01-05:
> We could do that in a later work.

I've raised an issue at waf repository and Thomas just added a new option `--clear-failed` on master :-)

The link to the issue is https://github.com/waf-project/waf/issues/1678 .

We could maintain a branch with the latest release plus that one commit in our module until the next waf release. And then make `waf check` or the wrapper set that option `--clear-failed`. Could both of us be happy with that @lucasdemarchi? :-)

cmarcelo, 2016-01-05:
To be clear, how can I accomplish the following actions (after the planned changes)?

- Compile whatever necessary and run all tests (regardless previous results)
- Compile whatever necessary and re-run any tests that failed previously or were changed

guludo, 2016-01-05:
@cmarcelo My suggestions would be the following:

> Compile whatever necessary and run all tests (regardless previous results)

waf: `waf check --alltests`
make wrapper: `make check-all`

> Compile whatever necessary and re-run any tests that failed previously or were changed

waf: `waf check`
make wrapper: `make check`

lucasdemarchi, 2016-01-05:
@guludo LGTM. @cmarcelo ?

cmarcelo, 2016-01-05:
ACK.
​

closed 2016-03-23 by lucasdemarchi