ArduPilot/ardupilot #3401: Autotune sets PI ratio that is not used during testing and makes the copter unflyable

andyp1per, 2015-12-30:
Autotune sets the roll and pitch I term to 0.1xP during testing but 1.0xP when testing is complete. This means that the test is not actually testing the parameters that are finally set and at least in my case (small 250mm copter, high CG) results in an unflyable copter.
At a minimum it should be possible to disable this (i.e. keep 0.1xP) or allow the user to configure the final ratio. A better solution would be to tune the I term, but I have no idea whether this is possible.

lthall, 2016-01-10:
What stops you from reducing the I term after autotune finishes?

I would be interested to see the data logs of the Autotune and flight afterwards.

andyp1per, 2016-01-10:
Hi Leonard, logs here:
http://diydrones.com/forum/topics/copter-3-3-beta-testing?commentId=705844%3AComment%3A2147568
The copter is unflyable which rather defeats the object of autotune, no?

lthall, 2016-01-10:
Hi Andy,
We set I to react to persistent offsets in approximately a second or two. 
The I term is set to 0.1P during the tests to ensure it doesn't mess up the P term test but still slowly corrects for persistent offsets.
The autotune algorithm is not able to tune the I term because we don't have a measurable requirement that we can test and uniquely tune I. So unless someone can propose an algorithm that we can use to tune I in the context of P and D then we can't move forward.

The objective of Autotune is to provide an automatic tune for well setup copters that don't have significant higher order terms in their transfer functions or coupling between axis. This means Autotune works well for better than 95% of copters if done correctly. For copters that are outside of the norm the pilot is able to manually tune their copter just like every other autopilot.

I recomend this Issue be closed.

closed 2016-01-11 by rmackay9

andyp1per, 2016-01-14:
Hi Leonard, why set I at all? Surely the precautionary principle would indicate its better to keep it low and allow users to manually tune up rather than picking something that may cause issues?