ArduPilot/ardupilot #3403: new warning after AccelCalibrator

lucasdemarchi, 2015-12-30:
Hi @jschall 

After the AccelCalibrator got merged I'm seeing this warning:

```
../libraries/AP_AccelCal/AccelCalibrator.cpp: In constructor ‘AccelCalibrator::AccelCalibrator()’:
../libraries/AP_AccelCal/AccelCalibrator.cpp:34:64: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 _param_struct(*reinterpret_cast<struct param_t *>(&_param_array))
                                                                ^
../libraries/AP_AccelCal/AccelCalibrator.cpp: In member function ‘void AccelCalibrator::run_fit(uint8_t, float&)’:
../libraries/AP_AccelCal/AccelCalibrator.cpp:336:79: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     struct param_t &fit_param(*reinterpret_cast<struct param_t *>(&param_array));
                                                                               ^
```

I think we have some alternatives to fix this.

1) Using an union and normal array

```c++
union param_u {
    struct param_t s;
    float a[9];
} _param;
```

This way you don't need to make param_struct a reference param_array: they are already sharing the same location in memory. Then _param_struct.xxx becomes _param.s.xxx and so on.

2) Turn struct param_t into a class and allow to access members as an array

I feel (1) would be better. Thoughts?

tridge, 2015-12-30:
it is strange that we don't get that warning here:
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_core.cpp#L30
Lucas, can you spot the difference?

lucasdemarchi, 2016-01-01:
@tridge The only difference I can see is that the reference is inverted.

This use of reference + cast looks like a nasty workaround to access the same memory location by 2 means, something the language has support for through unions. See this: https://gist.github.com/lucasdemarchi/e09639f8c1bd8152b505

Isn't this more elegant? I can give it a try.

lucasdemarchi, 2016-01-01:
We can continue the discussions on #3408

closed 2016-01-01 by lucasdemarchi