ArduPilot/ardupilot #3409: gtest needs to work on travis and

lucasdemarchi, 2016-01-02:
Besides the need to crosscompile gtest to use the binary on PX4, travis runs on Ubuntu which doesn't ship the libgtest library. See http://askubuntu.com/questions/145887/why-no-library-files-installed-for-google-test/145913. The reason doesn't really convince me but there's little we can do about it. To solve this I'd leave gtest as a submodule.

lucasdemarchi, 2016-01-02:
See https://github.com/lucasdemarchi/ardupilot/tree/pr/make-check and https://travis-ci.org/lucasdemarchi/ardupilot/jobs/99645066

guludo, 2016-01-05:
@lucasdemarchi Thanks! I'll work on it.

lucasdemarchi, 2016-01-05:
@guludo our fork for a submodule is now done: https://github.com/diydrones/googletest

We probably want to use a released version (1.7.0 would be a great one). As for integration what we could do is:

 - Find gtest from system - if it meets our needs, then we use it
 - Otherwise we fallback to compiling it ourselves.

What do you think?

cmarcelo, 2016-01-05:
For the reference, the updated link for the FAQ pointed in the AskUbuntu site is https://github.com/google/googletest/blob/master/googletest/docs/FAQ.md#why-is-it-not-recommended-to-install-a-pre-compiled-copy-of-google-test-for-example-into-usrlocal

cmarcelo, 2016-01-05:
Also, consider skipping the sample CMake build system there and use waf facilities to build GoogleTest static library.

Their README and the way the sources are collected together imply that approach is fine https://github.com/google/googletest/tree/master/googletest

lucasdemarchi, 2016-01-05:
@cmarcelo that reason looks so weak. It means "don't ever link a c++ library". And we have several examples of this working on other projects.

cmarcelo, 2016-01-05:
>    Find gtest from system - if it meets our needs, then we use it
>   Otherwise we fallback to compiling it ourselves.
> What do you think?

I would skip the first step. If it's not recommended or common to have the library, and we are compiling it anyway, why look in the system?

lucasdemarchi, 2016-01-05:
I agree about skipping the cmake build system.

guludo, 2016-01-05:
> @cmarcelo that reason looks so weak. It means "don't ever link a c++ library". And we have several examples of this working on other projects.

I agree with @lucasdemarchi here

cmarcelo, 2016-01-05:
> @cmarcelo that reason looks so weak. It means "don't ever link a c++ library". And we have several examples of this working on other projects.

The library probably exposes more things in the headers (to enable inlining), making it less friendly to keep it compatible among versions.

lucasdemarchi, 2016-01-05:
> I would skip the first step. If it's not recommended or common to have the library, and we are compiling it anyway, why look in the system?

All linux distros that I looked at except Ubuntu have it. The reason was probably caused by a bug report they received and decided to use the hammer to solve it. So if at all to find out what's the problem, let's keep looking for the system library.

guludo, 2016-01-05:
About skipping the cmake build system: +1 while gtest keeps the build "straightforward", as they say.

guludo, 2016-01-05:
> The library probably exposes more things in the headers (to enable inlining), making it less friendly to keep it compatible among versions.

Hm... I vote for trying to use the system's installation (if present) for now. We'll have a ready solution if the worse happens :)

guludo, 2016-01-05:
> @guludo our fork for a submodule is now done: https://github.com/diydrones/googletest

Thanks :-)

cmarcelo, 2016-01-05:
> All linux distros that I looked at except Ubuntu have it.

As explained in the FAQ entry, they used to recommend having the installed library.

lucasdemarchi, 2018-06-04:
No one I know of is going to work on this.

closed 2018-06-04 by lucasdemarchi