ArduPilot/ardupilot #3414: Navio start crash

samirbonnet, 2016-01-04:
After this commit below (therefore all others........ but previous commits work)
[ppm input destruction Navio death commit](https://github.com/diydrones/ardupilot/commit/8cca0beba9e5ab83a376fee2e57e30911e33cfcf)

Navio board immediately crash on startup (all kernel types and distros tested)
(All other RPI based hardware should be offline by now but I have only Navio+/Pi2 to test)

There is thousands of Navio and other RPI based ardupilot users but maybe only one user of that BH-Hat. Adding pwm input to those non realtime boards does not seem like a critical feature or may should be done by a BCM Dma expert but not at the cost of disabling all rpi based linux autopilots.

Lastly I am truly concerned that this kind of thing can happen: adding such a big/useless/destructing/hard to spot/impossible to debug (without previous bcmdma work)
"feature" without reviews.
Dma ppm part should be "secured", it is functional and only few people can work on it. 
 
Associated pull request.
[BH-Hat support](https://github.com/diydrones/ardupilot/pull/3282)

lucasdemarchi, 2016-01-05:
@samirbonnet thanks for the report and sorry for the inconvenience. There's no need to say how much one board is used as opposed to another and that a feature should not be included. We do our best to include the features without breaking others, but bugs happen.

In the PR I explicitly told about my concern wrt PWM input: https://github.com/diydrones/ardupilot/pull/3282#issuecomment-164513190

We will figure what's happening and provide a fix soon. I don't have a navio, so if you can run on gdb and provide the backtrace it would be great.

lucasdemarchi, 2016-01-05:
@samirbonnet the only way I can test on navio is in a remote board... and there it's **not** crashing. But I think this is because of there's no PPM connected there. @staroselskii can you help here?

vmayoral, 2016-01-05:
Hi @lucasdemarchi,
I compiled a branch including that commit and gave it a try (Erle-Brain 2, no PPMSUM receiver connected though) and here's what i got:
```bash
erle@erle-brain-2 ~/test $ sudo ./ArduCopter.elf 
Raspberry Pi 2 with BCM2709!
Interrupted
erle@erle-brain-2 ~/test $ sudo ./ArduCopter.elf 
Raspberry Pi 2 with BCM2709!
Interrupted
```
Mmm odd so i just started debugging:
```
erle@erle-brain-2 ~/test $ sudo gdb ./ArduCopter.elf 
GNU gdb (Raspbian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./ArduCopter.elf...done.
(gdb) r
Starting program: /home/erle/test/ArduCopter.elf 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Raspberry Pi 2 with BCM2709!
[New Thread 0x76bae450 (LWP 2507)]
[New Thread 0x763ae450 (LWP 2508)]
[New Thread 0x75bae450 (LWP 2509)]
[New Thread 0x753ae450 (LWP 2510)]
[New Thread 0x74bae450 (LWP 2511)]

Init APM:Copter V3.4-dev (180359d6)

Free RAM: 262144
FW Ver: 120
----------------------------------------

load_all took 572us
0 0 0 DataFlash_File: buffer size=16384
?	Q}?HMC5843: Could not detect version
?3?Calibrating barometer?%?	Q? ?3?Barometer calibration completePInit Gyro*?	Q?.**

Ready to FLY ??B%???????????'k???%??lD?p?<;
                                           ??? ? ? [du????O??	*???
'???
    >???H?
)|??????#??T??J????>8ʿ?RU?????F?ؤ?J
?[?\?P??>?Bʿ?<?:z>(:?ħ??#??JBh?</??<?=ox??lB?]?>?6ʿ?v?9?????V??^??J
?F??J%?????????&l?~? +?=lD?(<?=J
!??)??#??Z>+?*?~?              k??!? ? ? ?e?dl??"???F??#*q:?$>???}?%$??J77775e6?&#+???'+3i?Ҟ>?Qʿv3Z??L?????????(J
???5??s?"??>lʿj;2??????.???4J?V????2J??J???<:?<?f=?%?.??n?4>Wʿ4??:?5:e???N?/J
S??7:i???	8Qs??9?R$???????%jFQ?:5?>lD????>
?!?D????>Py?=???E??z?                           ?4?;?`?`?`?d?d\c?<?????=*?	?>'????>????@$??R77775?F?A#5??B5̍󾤋?>?yʿ@?Ǹ??h?'??R?CJ
b?S?Z$?????#k??T;y?lD`<?:?m?RJ2?S9?PJ??H?y? I?E?RW?<???<?J?<???J??󾸉?>fʿ???.L??]?8???KJ
?X?]??gW>               ???U?`?`?`?e?d1??V???m?W*KD?X>??_??Y$??Z77775??Z#;?c?[;??l??>s?ʿa?::?p?{p&:??\J
??<??^?ن?
???f     _;?/?`????????????a???? b?a?Z??<?,?<j2=??c!?4i???d???Q??>??ʿ*??:?]:ցȹ_??eJ
??󾃐?>
}??l!?1i^"?>??ʿ?mQ??^C?i????kJ
Program received signal SIGINT, Interrupt.
0x76f87c04 in write () at ../sysdeps/unix/syscall-template.S:81
81	../sysdeps/unix/syscall-template.S: No such file or directory.
```
Odd enough i was not able to reproduce the crash with `gdb`. Retried to launch the binary:
```
erle@erle-brain-2 ~ $ sudo ./ArduCopter.elf 
Raspberry Pi 2 with BCM2709!

Init APM:Copter V3.4-dev (03ec5bd3)

Free RAM: 262144
FW Ver: 120
----------------------------------------

load_all took 555us
0 0 0 ?	Q}?HMC5843: Could not detect version
No Compass backends available
COMPASS INIT ERROR
?3?Calibrating barometer?%?	Q? ?3?Barometer calibration completePInit Gyro*?	Q?.**

Ready to FLY ?B$?????????tJ???=lD33?;
                                     ???? ? 
                                            ? ?d????O??	*???
'???
    >?
$B7777?C?#??$??????>QvY??ߺS%????????J????????D??J??
                                                   ??r????? ?B?@?<?E?<~g=?
?S??$?>????/??"_{:?Jm?kT?J?Interrupted
```
Worked fine so it seems this issue is non deterministic. After i while i managed to caught it in `gdb`:
```
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/erle/test/ArduCopter.elf 

Program received signal SIGSEGV, Segmentation fault.
0x76fe597c in _dl_sysdep_start (start_argptr=0x7efff7d0, dl_main=0x76fd0328 <dl_main>) at ../elf/dl-sysdep.c:111
111	../elf/dl-sysdep.c: No such file or directory.
(gdb) bt
#0  0x76fe597c in _dl_sysdep_start (start_argptr=0x7efff7d0, dl_main=0x76fd0328 <dl_main>) at ../elf/dl-sysdep.c:111
#1  0x00000000 in ?? ()
```

It seems this issue also affects Erle-Brain 2

vmayoral, 2016-01-05:
Finished testing a rebased branch that removed the commit mentioned by @samirbonnet. With that removed, i have not been able to reproduce the crash. So i can confirm that there's an issue there.

lucasdemarchi, 2016-01-05:
@vmayoral do you have a board in the network that I can play with to test this?

lucasdemarchi, 2016-01-05:
@onceme you were the one doing this commit... if you can help debug it would be good.

onceme, 2016-01-06:
@lucasdemarchi let me check this issue and get back to you. Really sorry that if we would cause this trouble. Does it happen on other PPMSUM input source? I don't have a PPMSUM encoder, but I guess I would buy one today for debugging this issue.

Could you also inform the original author of RCInput_RPI to re-check the changes?

onceme, 2016-01-06:
I've purchased a PPM encoder module online, it would take 2~3 days on the road.

Till now writing this, I have tested RCInput_RPI in the latest master with BH + RPI2 like 20 times, but can't reproduce the crash.

My method is to change RCInput_RPI a little, so BH could use PPMSUM instead of separate PWMs. The changes are in this patch [bh_ppmsum_input.patch.txt](https://github.com/diydrones/ardupilot/files/79542/bh_ppmsum_input.patch.txt).

With the above patch, BH will use PPMSUM like NAVIO+ or ERLE board as rcinput and I could still do
```shell
$ make bhat -j4
```

The RPI_GPIO_4 is used by PCA9685 on BH, so change it to RPI_GPIO_5 in the patch.
```c
static uint16_t RcChnGpioTbl[RCIN_RPI_CHN_NUM] = {
-    RPI_GPIO_4
+    RPI_GPIO_5
};
```

This was the result without connecting actual PPM module
```shell
bit@bithollow:~$ sudo ./ArduCopter.elf
Raspberry Pi 2 with BCM2709!

Init APM:Copter V3.4-dev (ea823a81)

Free RAM: 262144
FW Ver: 120
----------------------------------------

load_all took 35us
0 0 0 DataFlash_File: buffer size=16384
�       Q}�HMC5843: Could not detect version
Failed to boot MPU6000 5 times
�3�Calibrating barometer�%� Q� �     QC^�3�Barometer calibration complete��Init Gyro**�   Q.PMinicom2.4*

Ready to FLY  ublox �   Q_�     Q�!EKF2 IMU0 tilt alignment complete
EKF2 IMU0 yaw alignment complete
�      Q�         Q���
Q�C^CInterrupted
bit@bithollow:~$ uname -a
Linux bithollow 4.1.9-rt8-v7+ #1 SMP PREEMPT RT Thu Oct 8 08:42:44 UTC 2015 armv7l GNU/Linux
bit@bithollow:~$ 
```

onceme, 2016-01-06:
Seems the original author is @vmayoral.

Hi @vmayoral, first I'd like to say thank you for the RCInput_RPI, which benefits many RPI based caps, BH is one of the them. While I was modifying its code, I checked almost every piece of your in-code comment, also read the BCM2835-ARM-Peripherals.pdf as well.

As we can see in this commit https://github.com/lucasdemarchi/ardupilot/commit/83e12465c4780d53ada08c3fca07bdbd80ac9e4f, my purpose is to support PWM inputs from several GPIO pins (thus can save ppm encoder hw). The original code just read the first byte (only RPI_GPIO_4 is required) of pin state registers via DMA, but now it reads 8 bytes for all the pins.

Because of this, I calculated the buffers again, updated all your related comments. Could you take a look at the code in detail to see if I missed anything? Or will the buffer size changing cause this  non-deterministic issue? I don't have either the NAVIO+ or Erle-Brain 2, and have not reproduced the issue on BH so far. Really appreciate.

onceme, 2016-01-06:
@lucasdemarchi Hi, before I receive the new PPM encoder module, please consider this commit https://github.com/bithollow/ardupilot/commit/aa7f5f520c8758494eab0331ddd27f6b4e348c7e as emergent fix. If necessary, please ask me to launch the PR.

It first rollback my changes, then adds _BH board #ifdef to use RPI_GPIO_5 as PPMSUM input.

We flied many times these days with BH v0.2, never met problem because of rcinput. I'm not sure if this has something to do with our Linux distro.

```shell
bit@bithollow:~$ uname -a
Linux bithollow 4.1.9-rt8-v7+ #1 SMP PREEMPT RT Thu Oct 8 08:42:44 UTC 2015 armv7l GNU/Linux
```

Our rootfs is based on Debian Jessie armhf. Seems most of the RPI user is still using raspbian.

samirbonnet, 2016-01-06:
I tried to be as concise as possible while indicating the exact location of the regression but as I said 
I tried different kernels up to 4.1.5-rt (never had issues with apm code so I tried gdb and distros/kernel first) and as I got gdb heisenbug without any change in the threading models just after a DMA related commit I was sure the issue was there so I posted the problem. 
Thanks !

Clouded, 2016-01-07:
@samirbonnet Thanks for raising this issue, this is shocking actually. 

@onceme So you didn't even test that PPMSUM Input works and submitted a PR? This is totally unacceptable. DMA is a very hacky and gentle thing and should be thoroughly tested even with minor changes. BTW, RCInput_RPI was originally written by Emlid.

lucasdemarchi, 2016-01-07:
@Clouded he tested on one of the boards since he doesn't have yours. Let's stop the bashing, please? I can raise problems with several commits, including from emlid, in which support for other boards has been broken and then we needed to fix it up on top. This is bleeding edge on master branch and instead of bashing we should be looking into the solution. BTW, he already sent a commit to fix it.

Clouded, 2016-01-07:
@lucasdemarchi Sorry if I seemed rude. No bashing here, my main worry is that (as far as I understood from the messages) new code was not tested with PPMSUM signal source, which obviously should have been done. 

@onceme When changing timings or block sizes please always check the performance as they strongly affect it. We've spent a lot of time finding the proper values.

lucasdemarchi, 2016-01-07:
@onceme thanks for the patch. I decided to take another approach: I reverted the original commit and then added the ifdef for GPIO5 in BH hat.

@samirbonnet the revert is now pushed to the master branch. Could you check it fixes the issue for you?

Clouded, 2016-01-07:
@lucasdemarchi Tested a803cfd1e881b0509e42ecfaece45f89437590bf on RPi2 with Navio+: no crashes, PPMSUM input works and performance is good.

lucasdemarchi, 2016-01-07:
@Clouded thanks. I think we can close this issue.

@onceme once you have it working with PWM and PPM please submit a PR and we will get it tested more.

closed 2016-01-07 by lucasdemarchi

Clouded, 2016-01-07:
@onceme When you have a new code I can test it on Navio+, just send me an email to mikhail.avkhimenia@emlid.com

onceme, 2016-01-08:
Hey guys, again, sorry for the trouble. My PPM encoder would arrive today or tomorrow, this time I will verify with real PPMSUM input before launch PR.

> to be honest I think this part of code should be the simplest possible for speed and understanding and after all ppm and pwm are different things !

@samirbonnet Hi, IMHO from the RCInput_RPI point of view, it is not aware of the pulse type (PWM/PPM/others), just pulse, which calculated from the elapsed time of GPIO level changing. Specific pulse type will be identified in RCInput. My modification (had been reverted) changes the data that transferring via DMA from only one GPIO level state to all GPIO level states; Also, it reduced the iterations of sampling, reduced the buffer size. Thus, I think, the patch would give RCInput_RPI higher performance and capable of sampling multi GPIO pins (useful feature, saving extra hw). Unfortunately, my patch was not well tested on the other Linux boards, this was my fault. I would like to spot the issue, fix, merge it again to master branch.

> once you have it working with PWM and PPM please submit a PR and we will get it tested more.

@lucasdemarchi Hi, thanks a lot for dealing with this issue. Sure, I will launch another RP once finished.

> When you have a new code I can test it on Navio+, just send me an email to mikhail.avkhimenia@emlid.com

@Clouded Hi, thanks. Probably I would ask you question about RCInput_RPI, frankly speaking, yes, the code is hacky, I might misunderstood something. Appreciate the contributions from Emlid;-)

samirbonnet, 2016-01-08:
@onceme thanks, you can contact me to check against navio target. got one !