ArduPilot/ardupilot #3422: Make windows FlightGear batch files

hamishwillee, 2016-01-05:
fg_plane_view.bat	 and fg_quad_view.bat	 are currently hard-coded to use FlightGear 3.4.0 based on the file location. It would be better to have these use an environment variable for the file location (IFF defined - e.g. FG_BIN) and otherwise test for possible known locations - using the most recent that is defined.

OXINARF, 2017-06-02:
With 59e4e8def6dd296067996813570b40de2553b920 the path is no longer hard-coded to 3.4.0 but still looks for a `C:\Program Files"\FlightGear*` path and doesn't allow the use of an env var.

@hamishwillee Do you think this still deserves to be open? I think most Windows users will install to default location and if they don't then they probably have enough knowledge to change the .bat files to point at their install location.

hamishwillee, 2017-06-02:
I think this can be closed. I just didn't like the hard coding to an out of date version of FG. The solution above is probably more elegant but not worth the effort.

closed 2017-06-02 by hamishwillee

peterchouuuuu, 2017-11-28:
ok,I have setup SITL on my PC successfully,but when look at the data of pitch when the copter is flying along  a straight line, I found that the data of pitch is wrong because that when flying along a line, the roll is zero,however the pitch is not zero ,which means the copter has a accerlation now.