ArduPilot/ardupilot #3437: Pixhawk won't enter ESC calibration

WaltHP, 2016-01-06:
I have bound my TX and RX.  I'm using ppm from the pixhawk to the rx.   

I have installed the latest qgroundcontrol on my mac.
I used qgroundcontrol in stall the latest firmware.

```
  Version: 4
  Board ID: 9
  Flash size: 2080768
Downloading firmware...
 From: http://px4-travis.s3.amazonaws.com/Firmware/stable/px4fmu-v2_default.px4
```

Steps:
I unplug the pixhawk
put throttle to max
plug pixhawk back in and wait for the musical tones, then unplug.
plug the pixhawk back int and the musical tones come up and I do not get the long tone.

I have also tried to use qgroundcontrol to do the esc calibration and that just fails instantly every time.
The app says "unplug battery and t"   I'm not sure what "t" means.

Steps:
plug the usb cable into the mac and connect with qgroundcontrol
hit the power tab
hit calibrate esc.  
Fails

```You must disconnect your battery prior to performing ESC Calibration.
Disconnect your battery and try again```

My battery was never plugged into the pixhawk.   I've also tried step 1 of plugging in the battery and following the same steps.  Same results.   I'm very frustrated at this point as I can't seem to do anything with it.

I'm at a total loss as what to do here.

magicrub, 2016-01-06:
Without a battery your ESC probably doesn't have power.

rmackay9, 2016-01-06:
This is a support issue so doesn't belong in github.  Sorry to be strict about this but if we're not our to-do list turns into a support forum.

closed 2016-01-06 by rmackay9

WaltHP, 2016-01-06:
If you notice, I also stated that I tried the same process with having my battery plugged in.  I've tried several different combinations and permutations of steps, and cannot get the pixhawk to go into esc calibration mode at all.  it just errors out using qgroundcontrol with no useful remedy.

dssoup, 2016-11-06:
Did the issue raised by WaltHP ever get resolved?

vibgyor-s, 2024-03-06:
Is this issue still active? Even I faced this issue recently.
ran on ardupilot just to make sure all ESCs are calibrated.

rmackay9, 2024-03-07:
@vibgyor-s,

I don't think there are any known issues with the ESC calibration code in AP.  If you're facing problems it is probably best to post in the support forums.  [Copter-4.4 forums are here](https://discuss.ardupilot.org/c/arducopter/copter-44/169) and [Copter-4.5 forums are here ](https://discuss.ardupilot.org/c/arducopter/copter-45/174).