ArduPilot/ardupilot #3456: Detect mid-flight motor/ESC failures

magicrub, 2016-01-07:
When throttle changes (goes up) but battery current stays the same, then we know the motor is not drawing current and a stall/crash will be inevitable. At this point we could go into a special fail-safe mode such as:
- loiter-down to ground
- with STALL_PREVENTION=0 do RTL with an enforced minimum sink rate of... > 2m/s? param?
- with STALL_PREVENTION=1 do RTL and let the stall prevention do the work of keeping us in the air, perhaps with an enforced minimum sink rate of >= 0? param?

Otherwise we will inevitably stall and fall out of the sky uncontrollably.

squilter, 2016-01-07:
Should this be part of the crash detector?

magicrub, 2016-01-07:
Failsafe is a better place

SlipstreamFPV, 2016-02-21:
Great feature idea and simple implementation. Only issue could be a failure of the current sensor triggering a false detection.

WickedShell, 2016-02-21:
So I have a log from someone who took off in manual, never armed the plane,
handed over to the autopilot (which of course not output throttle).
STALL_PREVENTION wasn't turned on, the plane basically just maintained FBW
min airspeed, quite successfully, which is actually a great behavior IMO.
(This only worked as the minimum airspeed is higher then the stall which is
why the plane still turned fine autonomously). This was with speed
weighting of 1, so I think the current behavior is far better then might be
expected.

proficnc, 2016-02-21:
Also Concider the possibility of a lost propeller.  Current will still rise, but not to the expected level.

magicrub, 2016-02-21:
I'm open to suggestions

SlipstreamFPV, 2016-02-21:
Current and voltage drop combined indicate a functioning motor. Insignificant voltage drop and no change in current = motor fail. Maybe even integrate an airspeed check since that should change with increased throttle as well.

magicrub, 2016-02-21:
At least in auto modes, throttle is only used to maintain your cruise airspeed. There shouldn't be any "increase" to look for. However, you make a good point about the voltage. During a flight your battery voltage is fairly noisy and bumpy. Going up and down by 10% or so for medium-high throttle vs none. That's another good indicator. It should have some deltas in it when the motor spins up.

summary:
- voltage should fluctuate and have delta spikes of 5-10% or at least whenever throttle goes from 0 to something. Need to ensure servos don't cause those spikes.
- throttle 40%+ should be drawing > 3A
- lost propeller is hard to detect, lets leave that for v2
- if STALL_PREVENTION is having to kick in during AUTO

jcooper21, 2016-03-14:
I think when the motor failure is detected ideally you would point the plane at the home position and attempt to glide home.

magicrub, 2016-03-14:
This feature will work best with a selector for the desired behavior.

marcmerlin, 2017-07-11:
For FPV pilots, the OSD should give a warning too. I'll look at writing some code to do this:
https://github.com/marcmerlin/night_ghost_minimosd-extra-mavlink/issues/1

iskess, 2017-07-11:
Several years ago actually had the opposite problem. I had my ESC stuck wide open and couldn't get the plane down and climbed into a rain storm. I had to wait until the battery died then dead-stick it in.  
At about 1:50 you can see the motor output dies with high throttle. Then later the throttle shows 0, but the motor is screaming.  
https://youtu.be/aANtCjEFzpA

rmackay9, 2019-08-03:
There is a corresponding issue here for Copter: https://github.com/ArduPilot/ardupilot/issues/11947

Naterater, 2021-07-26:
First, a windmilling prop at low-to-mid throttle will draw almost zero current, so I think this behavior could be falesly dangerous. In additon, I think we should be able to do this with scripting, correct? Can this issue be closed?

marcmerlin, 2021-07-26:
why would this be dangerous @Naterater ?
If you are windmilling, you also have close to 0 throttle output, so they match
The problem is if you have >50% throttle and you notice that you're using less than 3A, or whatever the correct value for a given aircraft.
I would love a control that takes "minimum amps at over 50% throttle", and if the amps dip below that, AP commands an immediate RTL as well as a mavlink message about motor failure

Naterater, 2021-07-26:
With 12x12 props, I can be in an updraft, flying along at 20m/s, 60% throttle, and <1A off draw when normally in level flight I’m at 40% throttle, and 4A. I’m saying the given conditions for “motor/ESC failure” here are too common to warrant specific failsafe coding in ardupilot. Furthermore, I believe you can already do this specific action as a part of scripting.

geofrancis, 2021-09-13:
wouldn't this be an ideal candidate for ESC telemetry? prop failure could be done by setting a maximum cruise propeller speed, motor/ESC/wiring failure would also be detectable by the low/no rpm from the motor or loss of ESC telemetry.

magicrub, 2021-09-13:
> wouldn't this be an ideal candidate for ESC telemetry? prop failure could be done by checking maximum cruise propeller speed so if a propeller came off you would see the motor is spinning faster than it would normally. motor/ESC/wiring failure could also be detectable by the low/no rpm from the motor or loss of ESC telemetry. esc telemetry also gives current and temperature readings from all escs individually so their heath can be monitored and problem motor disabled automatically.

Yup! Esc feedback is best. I think my intent here was to try and detect it with that but this is an old issue before we had esc telemetry. Now we just need to do more than just log/stream the data.. like declare an emergency or something

geofrancis, 2021-09-13:
> Yup! Esc feedback is best. I think my intent here was to try and detect it with that but this is an old issue before we had esc telemetry. Now we just need to do more than just log/stream the data.. [...]

yes something like propulsion failsafe, for single engine the options i would think about would be to limit bank angles and prioritize maintaining airspeed over altitude and start to RTL or closest rally point or landing waypoint.

you would also have to look at multi engine aircraft, as they would now be dealing with asymmetric thrust, so are going to have to keep a much higher minimum airspeed in order to reduce the risk of rolling over, they also need rudder input to fly straight or risk of rolling over. Something like rudder RPM compensation, where the rudder deflection is based on the difference in speed of the 2 motors.

then your into the VTOL aircraft, where you have a lot more motors running. a check that all the motors required for a flight mode work when the mode is changed would be needed. so when changing to Q loiter from forward flight, the flight controller should be sure that all the motors are actually turning before trying to vertical land.

andyp1per, 2022-04-10:
Yes, doing this with ESC telemetry is very easy

julled, 2022-12-22:
Very nice, i would be very interested in this feature!