ArduPilot/ardupilot #3461: Waf : Unable to use gbenchmark on ubuntu 15.10

khancyr, 2016-01-08:
Hi,

There is an issue with gbenchmark with waf on Ubuntu 15.10

````
[1652/1652] Linking build/linux/libraries/AP_Math/benchmarks/benchmark_matrix.linux
/home/khancyr/Workspace/APM/ardupilot/build/linux/gbenchmark/lib/libbenchmark.a(benchmark.cc.o): dans la fonction « benchmark::RunSpecifiedBenchmarks(benchmark::BenchmarkReporter*) »:
benchmark.cc:(.text+0x2e21): référence indéfinie vers « pthread_create »
/usr/bin/ld generated: référence indéfinie vers « pthread_create »
collect2: error: ld returned 1 exit status

Waf: Leaving directory `/home/khancyr/Workspace/APM/ardupilot/build/linux'
Build failed
 -> task in 'benchmark_matrix.linux' failed (exit status 1): 
    {task 139784788162576: cxxprogram benchmark_matrix.cpp.1.o -> benchmark_matrix.linux}
['/usr/lib/ccache/g++', '-Wl,--gc-sections', 'libraries/AP_Math/benchmarks/benchmark_matrix.cpp.1.o', '-o', '/home/khancyr/Workspace/APM/ardupilot/build/linux/libraries/AP_Math/benchmarks/benchmark_matrix.linux', '-Wl,-Bstatic', '-L.', '-lap', '-Wl,-Bdynamic', '-L/home/khancyr/Workspace/APM/ardupilot/build/linux/gbenchmark/lib', '-lm', '-lpthread', '-lrt', '-lbenchmark']
````

Using Using LDFLAGS instead of LIB in Tools/ardupilotwaf/boards.py make it work (see https://github.com/diydrones/ardupilot/pull/3460 )

Did other got the same issue ?

khancyr, 2016-01-11:
@guludo ,

I try to link gbenchmark with pthread in find_benchmark and it is working ( https://github.com/khancyr/ardupilot/commit/20b96e2bf2d04556b0fcb8ddbce38ac518f81d0e#diff-733f77c86a813ae61b6ab1149c151f9dL50 ) 
It pollutes the build with two -lpthread .... but I don't find another solution... I don't understand why waf is messing with linking ...

````
14:06:15 runner_env kw={'shell': False, 'cwd': '/home/khancyr/Workspace/ardupilot/build/linux', 'env': None}
14:06:15 envhash 23a19e68ae9c44db2049323e361398a8 [[], [], [], ['-o'], [], [], [], [], [], ['m', 'pthread', 'rt', 'benchmark', 'pthread'], ['/home/khancyr/Workspace/ardupilot/build/linux/gbenchmark/lib'], '-L%s', '-l%s', [], ['-Wl,--gc-sections'], ['/usr/lib/ccache/g++'], [], '-Wl,-rpath,%s', '-Wl,-Bdynamic', ['ap'], ['.'], '-L%s', '-Wl,-Bstatic', '-l%s']
14:06:15 task task  	{task 140656066218000: cxxprogram benchmark_matrix.cpp.1.o -> benchmark_matrix.linux} must run as the output nodes do not exist
[1683/1683] Linking build/linux/libraries/AP_Math/benchmarks/benchmark_matrix.linux
14:06:15 runner ['/usr/lib/ccache/g++', '-Wl,--gc-sections', 'libraries/AP_Math/benchmarks/benchmark_matrix.cpp.1.o', '-o', '/home/khancyr/Workspace/ardupilot/build/linux/libraries/AP_Math/benchmarks/benchmark_matrix.linux', '-Wl,-Bstatic', '-L.', '-lap', '-Wl,-Bdynamic', '-L/home/khancyr/Workspace/ardupilot/build/linux/gbenchmark/lib', '-lm', '-lpthread', '-lrt', '-lbenchmark', '-lpthread']
14:06:15 runner_env kw={'shell': False, 'cwd': '/home/khancyr/Workspace/ardupilot/build/linux', 'env': None}
Waf: Leaving directory `/home/khancyr/Workspace/ardupilot/build/linux'
Build commands will be stored in build/linux/compile_commands.json
'build' finished successfully (1.161s)
````

Another solution could be to use 
````
bld.env.LINKFLAGS.append('-pthread')
````

What do you think?

guludo, 2016-01-11:
Hi @khancyr

> I try to link gbenchmark with pthread in find_benchmark and it is working ( khancyr@20b96e2#diff-733f77c86a813ae61b6ab1149c151f9dL50 ) 
It pollutes the build with two -lpthread .... but I don't find another solution... I don't understand why waf is messing with linking ...

Indeed. By what I've seen in the source, it seems that waf doesn't have a mechanism of reordering and cleaning up the list of libraries. However, by looking at ld's man page:

> See the -( option for a way to force the linker to search archives multiple times.
> 
> You may list the same archive multiple times on the command line.

, it should be fine to have libraries being specified multiple times :-)

@lucasdemarchi any opinion on this?

guludo, 2016-01-12:
@khancyr I talked directly to @lucasdemarchi .

We like this:
```diff
-    env.LIB_GBENCHMARK = ['benchmark']
+    env.LIB_GBENCHMARK = ['benchmark','pthread']
```

Are you up for creating the pull request? :-)

khancyr, 2016-01-12:
@guludo 
Ok, I submit the pull request !

khancyr, 2016-01-14:
@lucasdemarchi @guludo 
After seeing your comment in https://github.com/diydrones/ardupilot/pull/3498 we can remove my fix and use -pthread instead of -lpthread by using LINKFLAGS  (https://github.com/khancyr/ardupilot/commit/3ac682e2fb65c6348ef9f64cab4c3880312cdb41 )
But ... -pthread seems related to gcc, can we use it in OSX or windows (with mingw ? )

guludo, 2016-01-14:
@khancyr I haven't tested waf build system on Windows yet, but the flag should be the same, since the target still is linux.

lucasdemarchi, 2016-01-14:
If it's compatible with both clang and gcc, we should go with that. Later we can detect in the build system what the proper flag for the compiler and use that. Like autotools do (and waf in samba probably does the detection already)

lucasdemarchi, 2016-01-14:
@khancyr can you open a PR?

lucasdemarchi, 2016-01-17:
Fixed.

closed 2016-01-17 by lucasdemarchi