ArduPilot/ardupilot #3466: ac3.3.2.rc2 bug - copter issues LAND command instead of RTL

imrj, 2016-01-09:
doing some testing via GCS, triggered GCS failsafe several times and it issued RTL command as expected and configured to do, however on one GCS failsafe it randomly issued a LAND command which was never configured to do. 
Yes,  RC Radio failsafe was off during all testing,  however this does not explain why previous GCS failsafe conditions were handled with RTL as expected but last one was a LAND instead where all parameters were the same all along 

, this is some sort of bug or inconsistent behavior of the software, log attached....thank you

imrj, 2016-01-09:
Static

imrj, 2016-01-09:
http://1drv.ms/1ZcPfpL

rmackay9, 2016-01-11:
If it's unable to do an RTL it will do a land.  So for example, bad GPS could cause that.  Also if it's very close to home (2m) it will trigger a Land.  Thanks for the logs.  I'm going to close this here, we can handle it on the Copter-3.3 beta testing thread.

closed 2016-01-11 by rmackay9