ArduPilot/ardupilot #3473: Plane: parachute deployment doesn't disarm

squilter, 2016-01-11:
@tridge 
I want to run a mission that ends with a parachute landing. Right now, a parachute deployment does not disarm the throttle. So the parachute would deploy, and then hopefully the crash detector would hopefully disarm a few seconds later. This is not ideal (although it is nice that it lets me set a backup behavior in case the parachute fails to deploy).  Here are some of my ideas to fix this:

1. Make every parachute deployment disarm the throttle. (optionally, use param2 to specify)

2. Support `MAV_CMD_COMPONENT_ARM_DISARM` in auto mode

3. Support `MAV_CMD_DO_FLIGHTTERMINATION` in auto mode (basically the same thing)

magicrub, 2016-01-11:
The Plane parachute release code disables the throttle for
auto_throttle_modes such as AUTO here
<https://github.com/diydrones/ardupilot/blob/master/ArduPlane/Attitude.cpp#L571>.
Have you tried it? Is it not working?

I agree that disarm is better your throttle should be off with the code
as-is.

-TomP

squilter, 2016-01-11:
I only tried it in sitl.  In copter it disarms immediately.  Plane didn't.  My sitl plane kept flying at the same speed, so I assumed the prop was still spinning.  In hindsight, it probably kept flying because the parachute isn't modeled in sitl (so there's no drag), and it already had a lot of momentum. 

What is the reason for suppressing throttle, not disarming?
Also, is the crash checker disabled after the parachute deploys?

magicrub, 2016-01-11:
Crash detection is not disabled for parachute. CRASH_DETECT default is 0 so
right now it won't disarm you by default. With CRASH_DETECT=1 a chute
release will (eventually) trigger a disarm. However, it's not a "crash" so
it should probably have an exception in there to not disarm. Actually no,
the chute logic just needs to disarm us and that will turn off the crash
detection.

iskess, 2016-01-11:
What if the parachute fails to deploy, and you end up needing throttle to salvage a landing? In this case disarming wouldn't be the best. You should always be able to add throttle in Manual mode.

magicrub, 2016-01-11:
Can't ague with that!

closed 2016-01-19 by squilter