ArduPilot/ardupilot #3480: waf: output the command string instead of python list representation

guludo, 2016-01-11:
When doing verbose builds with `-vv`, the commands are output as a python list representation instead of a full string with the command. Do if one wants to run one or more of the commands, she needs to convert it to list by hand (e.g. `python -c "print(' '.join(<command-copied-from-terminal>))"`). That's an annoying inconvenience :-)

Investigate if waf does provide a way to output the command string instead of a list. If not, it may be worth a patch to upstream.

Naterater, 2019-01-07:
I understand the build system has undergone quite a few changes recently. Is this still an issue @guludo?

peterbarker, 2021-07-30:
Yes, this is still an issue.

We make these strings copy/pasteable in autotest.