ArduPilot/ardupilot #3483: Plane: add WP_RADIUS = -1

magicrub, 2016-01-12:
With WP_RADIUS = -1 we would calculate the optimum value given the distance to next waypoint. The optimum radius is half the distance to the following waypoint.

WickedShell, 2016-01-12:
I don't really get this, what happens if the waypoint I'm flying to is 2km away? Would I turn at 1km? I think I'm missing something.

magicrub, 2016-01-12:
yeah, that's the point of this. WP_RADIUS does not do what you think it does. We need an image (on the wiki) to describe what a small vs a large number here does. I've added URL support to @Description specifically for this confusion

WickedShell, 2016-01-12:
Is this mostly meant to be relevant to the line cross aspect where we generate a extended line?

From the wiki: "Defines the maximum distance from a waypoint that when crossed indicates the waypoint may be complete. To avoid the aircraft looping around the waypoint in case it misses by more than the WP_RADIUS an additional check is made to see if the aircraft has crossed a “finish line” passing through the waypoint and perpendicular to the flight path from the previous waypoint." 

Thus I would imagine if we made it half the distance we would turn at 1km if it was 2km away. The bit after that is what makes it confusing :)

Naterater, 2019-01-04:
I don't think the current WP_Radius needs changing. When set to approximately the turn radius of the aircraft and L1 is tuned properly, the turns are excellent. Acute turns could use some work, but I think that should be a separate issue. If WP_RAD = -1 increases the radius for acute turns, then I'm all for this. Otherwise I don't really follow what the issue is. 90-degrees and obtuse turns turn quite well.