ArduPilot/ardupilot #3488: Gazebo simulator not working

harewei, 2016-01-12:
Hello

I have followed the instructions on this page
http://dev.ardupilot.com/wiki/using-rosgazebo-simulator-with-sitl/
the only thing I missed was I couldn't 
git checkout RangeFinderSITL2
and I couldn't use 
rosdep install --from-paths src --ignore-src --rosdistro indigo -y

For the rangefinder problem, I have ignored for now, and the second problem I followed someone's instruction from
https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/194
which says go to ~/catkin_ws/src and run rosdep install mavros

Now, I have managed to finish building with catkin_make, but when I call the function
sim_vehicle.sh -f arducopter_sitl_ros --console
all it shows is an XTerm problem with
Starting sketch 'ArduCopter'
Starting SITL input
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
and a console .

The image at the bottom of the page (the one with the copter in gazebo) never shows up.  I have also read 
http://dev.px4.io/simulation-gazebo.html
which gives a very similiar image, but never got that working too, as explained from my issue post here
https://github.com/PX4/Firmware/issues/3495

Any help appreciated!

lucasdemarchi, 2016-01-17:
@ricardotk do you have the steps you used handy?

ricardotk, 2016-01-18:
@harewei take a look in this [document](https://docs.google.com/document/d/1rPudWJhkHtMqwLn_B6OflWD1gyMWahyJkqx2e68Rl9w/edit?usp=sharing).

khancyr, 2016-01-18:
Erle guys made a nice explanation too : http://forum.erlerobotics.com/t/simulate-erle-copter-in-gazebo/645 (plus video )

harewei, 2016-01-19:
thank you @ricardotk and @khancyr I will look at those later as alternatives.  right now I managed to get the simulator working with gazebo with http://dev.px4.io/simulation-gazebo.html

closed 2016-10-29 by squilter