ArduPilot/ardupilot #3509: AP_HAL_Linux : Navio crash

samirbonnet, 2016-01-15:
Sorry to raise this issue again. 
Navio immediately start crash "Interrupted", if someone else have this issue 
$ git clone master and make -j4 navio. Then confirm.

again commit [BH: Support UBLOX GPS M8N(https://github.com/diydrones/ardupilot/commit/800c142ac0e562510064668aa83a9ad35504cdf1)
works perfectly.

sudo su
gdb --arg ./ArduPlane.elf -A .......... works as usual with master (don't know for ppm however, will test it as soon as I get back my remote, but btw running a process through gdb : not fan)

While testing make sure to get a perfectly clean repo, no overclock, no binaries in /tmp folder.

Thank you again.

vmayoral, 2016-01-17:
Thanks for reporting @samirbonnet.

@lucasdemarchi this issue also affects Erle-Brain 2 and the PXFmini.

lucasdemarchi, 2016-01-17:
@samirbonnet could you test these 2 commits?
 - https://github.com/diydrones/ardupilot/commit/f449107563a7ab6444d9002c273c66766eb304d9
 - https://github.com/diydrones/ardupilot/commit/321803919caee413ac0ba07204ac4bc89a5464f3

Posting the coredump and/or backtrace in gdb when it crashes.

Thanks for the report.

lucasdemarchi, 2016-01-17:
@vmayoral just got my Erle-Brain2... I'll take a look on it in the beginning of the weak if nobody beats me to it.

lucasdemarchi, 2016-01-17:
@samirbonnet additionally, what RC do you have connected? Just tested master but without an RC connected - not able to reproduce it.

lucasdemarchi, 2016-01-17:
@vmayoral do you mean you tested it and were able to reproduce the issue?

vmayoral, 2016-01-17:
@lucasdemarchi yes, gave it a try and got the same results @samirbonnet described (no RC connected in my tests).

Binary crashes. Launched the debugger and first time the autopilot run correctly but after re-running it in the same gdb session, it crashed. The debugger ends up failing so it could be gdb-related. 
Have a look at https://gist.github.com/vmayoral/5e96e3a270581601fb69.

samirbonnet, 2016-01-17:
commits 
https://github.com/diydrones/ardupilot/commit/f449107563a7ab6444d9002c273c66766eb304d9
https://github.com/diydrones/ardupilot/commit/321803919caee413ac0ba07204ac4bc89a5464f3

worked ok, ppm input ok.
Now will let https://github.com/diydrones/ardupilot/commit/f449107563a7ab6444d9002c273c66766eb304d9 run for some days.

samirbonnet, 2016-01-17:
https://github.com/diydrones/ardupilot/commit/f449107563a7ab6444d9002c273c66766eb304d9
Interrupted after sometimes. Trying https://github.com/diydrones/ardupilot/commit/321803919caee413ac0ba07204ac4bc89a5464f3

lucasdemarchi, 2016-01-17:
@samirbonnet any chance you have 2 instances of ardupilot running?

samirbonnet, 2016-01-17:
lol. no.

lucasdemarchi, 2016-01-18:
@samirbonnet @vmayoral 

So.. I just pushed some commits into master to revert the PWM input changes. @onceme sorry, but we will need to try again. This time I have an Erle-brain2 with me so I can help debug, but today I can't finish the debugging session.

I could not reproduce the issue in a RPi 2 + navio board from @tridge. But I can reproduce it very easily with the erlebrain2.  I'm not sure why.

@samirbonnet @vmayoral could you check master is working for you? Thanks.

lucasdemarchi, 2016-01-18:
btw... I also added an strsignal() call inside the signal handler, so at least now we know it's catching a SIGSEGV.

onceme, 2016-01-19:
> lol...
> f449107 is random interrupting while Master crash at start < (yes easier to see)
>
> and commit without ppm work as always 800c142.
> That's it.
>
> I am know testing 3218039 and no interruption happened yet.

@lucasdemarchi 
If @samirbonnet could reproduce the crash (interrupted by SIGSEGV in RCInput_RPI) with f449107 (up to this commit, master has no pwm changes as you knew), then he probably observed the same result in https://github.com/diydrones/ardupilot/pull/3489#issuecomment-170924995

@samirbonnet Hi, have you saved the coredump and/or backtrace? If so, could you please share it with us? Thanks.

lucasdemarchi, 2016-01-19:
@onceme since the rpi boards register a signal handler for all signal, any signal will end up reproducing a "Interrupted" panic message, no matter where it comes from. So it could be crashing elsewhere due to other bugs.

Yesterday I could reproduce 100% of the time in an erlebrain2 with the PWM input patches applied and 0% of the time with them reverted. That is without any receiver connected. I was working on master branch and master + reverted commits. Interestingly accessing a RPi + navio board remotely, I could run ardupilot without any issue.  I bet there's memory corruption somewhere (even not inside RCInput_RPI) that's causing this behavior.

As for gdb and backtrace: as soon as gdb was plugged in the crash couldn't be observed anymore.

onceme, 2016-01-19:
@lucasdemarchi 
> @onceme since the rpi boards register a signal handler for all signal, any signal will end up reproducing a "Interrupted" panic message, no matter where it comes from. So it could be crashing elsewhere due to other bugs.

Yes, I was going to tell you this in similar words:)

One more thing is that, on my side, once the ardupilot process started without crashing, it wouldn't crash (with/without pwm changes in RCInput_RPI) in my aging test. So I guess we can narrow it down to the startup phase of the process.

Since you can reproduce it 100% in erlebrain2, I found a little bit of difference between erlebrain2 and bh (probably navio), it's about the loading parameter time:

erlebrain2
```shell
Init APM:Copter V3.4-dev (180359d6)

Free RAM: 262144
FW Ver: 120
----------------------------------------

load_all took 572us
```

bhat
```shell
Raspberry Pi 2 with BCM2709!

Init APM:Copter V3.4-dev (ea823a81)

Free RAM: 262144
FW Ver: 120
----------------------------------------

load_all took 35us
```

navio
```shell
Init APM:Copter V3.4-dev (decb6dba)

Free RAM: 262144
FW Ver: 120
----------------------------------------

load_all took 74us
```

bh and navio load_all time were similar, but seems erlebrain2 took more time. Would this timing cause some different while starting up?

samirbonnet, 2016-01-19:
as @onceme pointed out, in some commits
Ardupilot start crash "Interrupted". Then on a second restart the binary works and even after a reboot.
For example https://github.com/diydrones/ardupilot/commit/321803919caee413ac0ba07204ac4bc89a5464f3 

While in some others it just fails everytime. Starting the process even for the second time will always fail.

I am using some fast 8 channels ppm generated signals for benchmark.
It is working great even on those "second start needing binaries".

I need more time to confirm this behavior. Please be patient.

samirbonnet, 2016-01-20:
I was not able to reproduce the bug. Even when running ArduPlane.elf for 12 hours.
Tried with no gps, no ppm, dumb overclocks and settings :D, different ground station, etc... Binary don't want to crash. The only thing that changed was rpi-firmware so maybe it is related. 

commits involved
https://github.com/diydrones/ardupilot/commit/f449107563a7ab6444d9002c273c66766eb304d9
https://github.com/diydrones/ardupilot/commit/321803919caee413ac0ba07204ac4bc89a5464f3

I can't even reproduce the behavior "have to be started twice".

However I think the idea of @lucasdemarchi to revert temporarily until things are more understood is a good idea because this was kind of a very unsafe and vicious bug.

Obviously master is working perfect. well... predictable...
PPM performance is also very good with my mean homemade generator, no regression.

And yes of course I still got an immediate startup crash "Interrupted" with the commit [AP_HAL_Linux: support PWM input for BH hat](https://github.com/diydrones/ardupilot/commit/8cca0beba9e5ab83a376fee2e57e30911e33cfcf)

samirbonnet, 2016-01-20:
I suggest to close this issue

lucasdemarchi, 2016-01-27:
@samirbonnet thanks for testing. We will let you know when we merge again the changes to do PWM and cause possible crashes.

closed 2016-01-27 by lucasdemarchi