ArduPilot/ardupilot #3513: AHRS pream fail status it not sent to MinimOSD

kosmikkosmik, 2016-01-16:
I believe there is a problem here:
    if (!position_ok()) {
        if (display_failure) {
            const char *reason = ahrs.prearm_failure_reason();
            if (reason) {
                GCS_MAVLINK::send_statustext_all(MAV_SEVERITY_CRITICAL, "PreArm: %s", reason);
            } else {
                gcs_send_text(MAV_SEVERITY_CRITICAL,"PreArm: Need 3D Fix");
            }
        }
        AP_Notify::flags.pre_arm_gps_check = false;
        return false;
    }

I do see "PreArm: Need 3D Fix" message delivered over MAVLink to MinimOSD, but not messages about e.g. GPS horizon error.

I've replaced GCS_MAVLINK::send_statustext_all with a custom method similar to gcs_send_text_fmt() and started to get messages correctly.

I don't have a pull request yet since I've make a custom version from Copter-3.3 branch.

rmackay9, 2016-01-18:
ok, looking forward to hearing more of what you've found.  I would have expected send_statustext_all would send the message to all connected serial ports but perhaps it's not for some reason.

kosmikkosmik, 2016-01-18:
Is there any "spec" about what is expected from GCS to be considered "active" from viewpoint of mavlink lib? Like sending specific messages, answering hearbeats? To me it definitely looks like Copter class and mavlink library have different views on "active" state of GCS.

rmackay9, 2016-01-19:
I think a heartbeat at 1hz is the minimum required for us to consider the GCS active.  What you're perhaps seeing is that copter doesn't trigger a GCS failsafe unless the rc-overrides are being used.  There's a to-do item to change this so that it activates when it loses the heartbeat.

kosmikkosmik, 2016-01-19:
If I add hearbeats to MinimOSD then the current code works, I do get broadcasts. However I wonder if it's really a proper way to go. Consider this situation:
* MinimOSD with heartbeats
* Real GCS

Now the "real" GCS is lost, am I right that APM will not initiate GCS failsafe since there is MinimOSD which is pulsing?

magicrub, 2016-01-19:
MinimOSD is supposed to work with one-way data, right? Sounds like that
model should be kept if possible instead of forcing everyone to rewire.

kosmikkosmik, 2016-01-21:
OK after playing some more.

The thing is that there are multiple MinimOSD-Extra versions here. Some of them request data streams, some of them do not (the method is just not called). I took the "Copter" one and it actually didn't request data streams at all. The interesting thing is that it still got data, but the actual piece of code which decides if GCS is active from viewpoint of library expects that at least one message is received. After fixing it in MinimOSD I started to get updates correctly. 

So if the requirement on Mavlink GCS is to at least send something it works as intended.

Saxin, 2017-03-08:
@kosmikkosmik Can you please ellaborate more what ver. ? (I guess it's night-ghost rework ?) 
I'm struggling with his version, sharing Tel. 2 on Pix. with esp8266 radio, I get some stuff to update + messages but most of the stuff don't update.

OXINARF, 2017-03-09:
@kosmikkosmik After reading the history of this issue it looks like the problem was that OSD wasn't requesting streams (which is how MAVLink works for most repeated info). For this reason I'm closing this, let me know if you think there's an ArduPilot issue still present.

closed 2017-03-09 by OXINARF