ArduPilot/ardupilot #3517: Copter: Arming /motor start  warning issue

AndKe, 2016-01-17:
To reproduce: 
observe normal motor-start warning:
throttle stick in low-right position, audible warning is played, then motors start.
now disarm (throttle stick in low-left) ,then start motors again
This time, the motors start, simultaneously with the warning beep.  (there is no real warning)

rmackay9, 2016-01-18:
Thanks for the report.  There's a PR that addresses this from @jschall  and @R-Lefebvre that needs to be reviewed/merged.

rmackay9, 2016-08-08:
Copter-3.4-rc2 (which should appear in the mission planner's beta firmwares link in about an hour) includes a 2 second delay after arming and before the motors start.  I think that resolves this issue so I'll close this now.

closed 2016-08-08 by rmackay9