ArduPilot/ardupilot #3519: Make Use of SBGC_lib From AlexMos GitHub Repo

davidbitton, 2016-01-18:
I was looking over the code for [AP_Mount_Alexmos](http://tro.pe/1Rw0ASF) and I noticed that the code does not take advantage of the [SBGC_lib](http://tro.pe/1RvZQwT) from alexmos's [SimpleBGC Serial API examples and libraries](http://tro.pe/1Rw0H0A) repo. Would it make sense to use their lib for defines et al? I'm willing to take on the rework if it's something the collective deems a worthy effort. Thanks!

rmackay9, 2016-01-19:
@davidbitton, I think that might make sense.  You'd use a git submodule to link it?  So it could appear in this directory.  https://github.com/diydrones/ardupilot/tree/master/modules

davidbitton, 2016-01-19:
Yes.

Please excuse its brevity and/or typographical errors.

davidbitton, 2016-01-20:
Which repo should I work from for my Pixhawk?

davidbitton, 2016-01-21:
I tried to add their code as a submodule and add it to the build process; too confusing. can you help with adding it into the Makefiles?

OXINARF, 2016-12-20:
@davidbitton Can we close this? I think that adding a dependency on a simple helper library doesn't bring much benefit to be honest. Also, we have a PR (#5297) that improves Alexmos support and brings new definitions in.

davidbitton, 2016-12-20:
done.

closed 2016-12-20 by davidbitton