ArduPilot/ardupilot #3534: enhancement : Considering switch to travis trusty

khancyr, 2016-01-22:
Hi,

I just made some test on travis trusty and it works building APM!
https://github.com/khancyr/ardupilot/tree/test_travis_trusty
https://travis-ci.org/khancyr/ardupilot/builds/104040447

With the current config, it doesn't seem to improve build speed, but it should we some tuning and using trusty package .

magicrub, 2016-01-22:
What's different about Travis trusty vs regular Travis?

khancyr, 2016-01-22:
Actually travis is running ubuntu 12.04 "precise" , switching to trusty should simplify ci a little by using up to date package (and over use apt-get ) and maybe build a little faster

lucasdemarchi, 2016-01-27:
@khancyr I don't think it will speed up the build, unless travis is giving preference to machines using it. Anyway, if it builds fine, we can switch.  Could you send a PR?

lucasdemarchi, 2016-01-27:
@khancyr btw, recently we added support to semaphoreci as well, which is currently building the code way faster than travis

closed 2016-11-09 by khancyr

magicrub, 2016-11-09:
Please provide a reason for closing.

khancyr, 2016-11-10:
Using trusty on travis is quite slower to start and the transition will not be that simple. So this proposal should be close for now until some made a PR for using trusty