ArduPilot/ardupilot #3536: Setpoint Position + Velocity

khancyr, 2016-01-22:
Hi,

I try to use guided mode in position but I want to control the uav Z speed. 

According to https://github.com/diydrones/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1597
I can send pos + vel order in SET_POSITION_TARGET_LOCAL_NED message. 
Send z pos and z velocity doesn't work ... only pos z is take into account and I can control z speed . Is that normal ?

humm in fact it works ...but strangely ... on x/y,  it works like that : 

newSpeed = speed (dont know where it is defined) + targetSpeed ;
(if speed + targetSpeed > speed, it does nothing )

On z axis, it do the same but I need to use extrem zspeedvalue like 10000 ...

Is there a way to control position and speed to reach position (I only use z axis ) 

By the way, according to wiki, it shouldn't work : 
````
At time of writing you must enable all three bits for the position (0b0000111111111000) OR all three bits for the velocity (0b0000111111000111). Setting just one bit of the position/velocity or mixing the bits is not supported ````
http://dev.ardupilot.com/wiki/copter-commands-in-guided-mode/#set_position_target_local_ned

OXINARF, 2017-06-02:
Is this still valid? Looking at current code I can't see why it wouldn't work.

Regarding documentation, it still needs to be updated, but I believe that is always a work-in-progress and it doesn't deserve an issue for just for this.

rmackay9, 2017-06-02:
I think @khancyr was perhaps misunderstanding how the pos-vel controller works.  It's easy to see how it could be confusing.

@khancyr expects that the caller can input a position and a velocity and the vehicle will proceed to that position at the given velocity.  That's not how it works though, what the caller should provide is a continuous stream of desired positions and a desired velocitys for the vehicle.  I.e. I want to be at position A travelling at velocity V.  The reason this kind of controller is useful is to allow the caller to make the vehicle fly on a specific curved track (like a circle).  At each moment the caller could provide an updated target position and velocity and the vehicle would do it's best to achieve both.

I'm going to close this item.

closed 2017-06-02 by rmackay9

khancyr, 2017-06-02:
I will retest but I thinks Randy explanation is right ! maybe it shoud be add as comments somewhere that a stream is needed and not just a single command like for setpoint position !