ArduPilot/ardupilot #3547: EKF2: allow multiple EKF instances to use multiple GPS instances

magicrub, 2016-01-25:
In a dual GPS and multiple instance EKF2, allow EKF2.instance1 to use GPS.instance1 and EKF2.instance2 to use GPS.instance2 and totally bypass the sat-count based GPS-primary logic.

proficnc, 2016-01-25:
Love it :)

tridge, 2016-01-27:
Paul has explained that the motivation was a GPS that didn't configure the correct navigation filter, leading to vertical velocity lag and thus poor EKF state estimation.
Adding the feature of having a separate GPS per EKF instance is fine, but I don't think we should consider this fixed until we also have an arming check that checks if all GPS instances have been correctly configured. The uBlox GPS driver knows if it has received an ACK from the GPS that it has been configured correctly (for message types, message rate and navigation filter), but at the moment this isn't used as part of an arming check or health report. We should fix that.
That will improve things for all users, even if they only have 1 GPS.

WickedShell, 2016-01-27:
Tom and I have discussed it and he has a closed pr that I'll be reworking
on the ublox side.

magicrub, 2016-01-27:
>>but I don't think we should consider this fixed until we also have an
arming check that checks if all GPS instances have been correctly configured

Funny you mention that, Tridge. See
https://github.com/diydrones/ardupilot/pull/3546

arnaudstephan, 2016-09-03:
Hi everyone, maybe the solution is hardware. Ublox release the m8u chipset this summer with the dead reckoning included. If gps failed, it uses an integrated imu and accelerometer to estimate the position. I'm wondering if this can work fine additionaly to the ekf filtre applied. Thank you.

magicrub, 2016-09-03:
not in our case because we do the same thing but with additional sensors so
you end up with a better result

arnaudstephan, 2016-09-03:
Ok i understand.

WickedShell, 2017-05-29:
@priseborough is there still any value in taking this approach given the blending support now in master?

magicrub, 2017-05-29:
I think this is still a wish list item. As computation resources open up the ability to do a 2gps x 3imu matrix becomes possible.

priseborough, 2017-05-29:
I agree, the matrix approach  it is something to aim for longer term because it enables use of median selection algorithms which are a proven way of rejecting uncorrelated errors.

IamPete1, 2021-08-30:
We can do this on EKF3 now, https://ardupilot.org/dev/docs/common-ek3-affinity-lane-switching.html

closed 2021-08-30 by IamPete1