ArduPilot/ardupilot #3549: LAND or RTL is started when YAW in PosHold..where to disable?

AmsterdamHolland, 2016-01-25:
Hi everyone, 

I am new to Mission Planner.
My quadcopter is flying OK, and all functionalities are working, but when I have it in Position Hold, and use Yaw to look around, almost immediately LAND or RTL is started ( not sure if it is LAND or RTL yet, have not flown since). 
I taketh there must be a setting in Mission Planner where I could disable the LAND or RTL, so that I can yaw/look around freely.

Anyone who can steer me in the right direction?

Thanks

rmackay9, 2016-01-26:
We will need to see a log file.  It's most likely that a failsafe is triggering, probably an EKF failsafe.

closed 2016-01-26 by rmackay9

AmsterdamHolland, 2016-01-28:
Hi I have the log file attached
[LogSettings.txt](https://github.com/diydrones/ardupilot/files/108363/LogSettings.txt)

I compared to another qaudcopter (the same, different owner) log file, the EKF is both set to 0,8, when turned to 0 it shouldn't Land? But I don't think this is wise, as the landing is not happening with this other quadcopter. It is the UPair Chase by the way. Thanks for the help.