ArduPilot/ardupilot #3553: Using of isnan function

approximator, 2016-01-26:
Hi,

File: https://github.com/diydrones/ardupilot/blob/master/libraries/AP_Math/quaternion.h#L53
Comment: https://github.com/diydrones/ardupilot/commit/c7d5f06b21fa4f4df53a414909bc477c481c2bad#commitcomment-15622602

I am wondering why isnan is used here instead of std::isnan? 
Now it can't be compiled with mingw32 compiler.

I understand that it is not supposed to be compiled with mingw, but I just curious if there is a reason to call isnan in a such way.

Thank you in advance.

lucasdemarchi, 2016-01-27:
Is this the only reason?

Some functions are missing on some toolchains, particularly the GCC 4.9.3 the PX4 port is using.  If you just replace with std::isnan and it works everywhere it could probably be used.

rpadovani, 2016-04-18:
I have the same problem with _ g++ 5.3.1_

There is anything I can do without having to change each file to `sdt::isnan()` ? (or `using std::isnan()`)?

OXINARF, 2016-04-18:
See #3878. We are deprecating the use of make build system for Linux, you should start using waf.

lucasdemarchi, 2016-04-18:
And
http://discuss.ardupilot.org/t/waf-to-become-default-build-system-for-linux/8451/8

OXINARF, 2016-05-03:
Closing this in favor of #4026

closed 2016-05-03 by OXINARF