ArduPilot/ardupilot #3555: Accel calibration does not wait for user input

rmackay9, 2016-01-27:
A user or two has reported this issue.  What happens is during the accel calibration it does not wait for the user to reposition the vehicle and press the "ok" button (or whatever) on the ground station.  Instead it just keeps progressing every second or so to the next step which inevitably leads to a calibration failure.
The issue is probably that the AP_InertialSensor_UserInteract_MAVLink::_snoop function is not discriminating enough. It seems any COMMAND_ACK will cause it to proceed.  It should probably check that the "result" field is "MAV_RESULT_ACCEPTED" and we should probably specify the "command" field should be "241" for MAV_CMD_PREFLIGHT_CALIBRATION.

https://github.com/diydrones/ardupilot/blob/master/libraries/AP_InertialSensor/AP_InertialSensor_UserInteract_MAVLink.cpp#L20

lucasdemarchi, 2016-01-27:
@rmackay9 I also noticed this behavior to be only reproduced on APM Planner. QGC and mavproxy work ok.

peterbarker, 2017-12-19:
@rmackay9 Any more reports?

WickedShell, 2018-03-10:
I'm closing this because it's seen no activity in quite awhile, and the MAVLink snooping approach should be considered deprecated and we should be encouraging GCS maintainers to use #5200 instead as it is a much, much more robust approach.

closed 2018-03-10 by WickedShell