ArduPilot/ardupilot #3556: Adding Code to Build Process

davidbitton, 2016-01-27:
I added additional code (SBGC library) under AP_Mount, but it's not being built. The modules.mk file in the root says it's auto generated. Where do I add a reference to the code in order for it to be built when performing a `make px4-v2`?

guludo, 2016-01-28:
@davidbitton could you point to the branch you've been working on?

davidbitton, 2016-01-28:
My [fork](https://github.com/davidbitton/ardupilot) of master. I was able to get the code to build by moving the SBGC_lib folder alongside AP_Mount, and not as a subfolder. Then I added `LIBRARIES += SBGC_lib`to the make.inc file under `ArduCopter`. The next step now is rewrite AP_Mount_AlexMos using the sample code @alexmos published.

guludo, 2016-01-28:
Your fork just has commits related to submodule and a merge commit.

davidbitton, 2016-01-28:
i started down the road of adding SBGC_lib as a submodule, but I became
confused quickly with the build process. I figured, let me get the code
working first, and then I'll worry about submodules later.

lucasdemarchi, 2016-01-31:
@davidbitton I think you figured out what's needed, right? Can this be closed?

davidbitton, 2016-01-31:
Yes, however I'll be back when it's time to add this as a us module.

Please excuse its brevity and/or typographical errors.

lucasdemarchi, 2016-01-31:
Ok, feel free to open another issue if it's needed when you are adding it as a submodule.

closed 2016-01-31 by lucasdemarchi