ArduPilot/ardupilot #3560: Option to prohibit switch out of auto modes with very low throttle

faceEater, 2016-01-28:
I had a crash with my F450 today, which, after looking through my logs, appears to have been caused by the following series of events: I am flying my quad in stabilize mode, and the battery gets low, tripping the low-voltage failsafe. This causes the quad to switch to RTL mode, and start going upwards to it's RTL height. I respond by pulling down on the throttle, trying to compensate for the upwards motion. I then quickly realize that RTL mode has been activated, so I flip the mode switch up and back to it's old position to disengage the failsafe. I failed to realize, however, that I had already lowered the throttle, so the quad falls out of the sky when stabilize mode is re-engaged. A software option to prohibit exiting RTL, auto, or loiter with throttle at or near 0 while in the  air could have prevented this crash, and seems like a reasonable option to provide. I could also see this scenario taking place if one wants to take over manual control of a flight that has so far been fully auto-mode, and forgets to raise the throttle.

peterbarker, 2016-01-28:
The senior-telemaster quadplane went down to a similar issue from guided mode.

Perhaps some sort of throttle slew would be more appropriate than barring the mode change, however.

faceEater, 2016-01-28:
I agree with your suggestion for limiting throttle slew. This does seem like a better solution.

R-Lefebvre, 2016-03-02:
Every time this issue comes up (and it's often) the point is raised that low throttle stabilize is also the method that a user would take to quickly kill the motors should the copter be heading towards a collision.  In fact, it used to be the ONLY sure-fire way to stop the motors quickly.  For this reason, we never supported anything that would prevent it's functioning.

This is why I created the Motor Interlock function.  It provides an absolute motors on/off function on a switch.  This is much safer.  And because of this, using this function eliminates the motor shut-down function in Stabilize.

I haven't heard anybody who tried the function, saying they ended up not liking it.  Even the harshest skeptics.

lthall, 2018-01-11:
Given the age of this issue I am closing it.

closed 2018-01-11 by lthall