ArduPilot/ardupilot #3561: Feature request: Multiple low battery thresholds

faceEater, 2016-01-28:
It would be convenient to have an option to sound the beeper to allow the pilot to take manual action at some voltage above that which triggers the battery failsafe. I don't want my copter to RTL/LAND every time it gets low on battery due to the risk of colliding with an obstacle while trying to do so, but it would be good if it could do so if its battery reached a "critical" level, after having started beeping when the battery was "low" (for example, this would be useful if I couldn't hear the beeper, or it were broken/unplugged). The possibility to make the failsafe choose LAND instead of RTL if it is too far from home would also be a nice touch.

kosmikkosmik, 2016-02-13:
Maybe a better way is to incorporate distance and RTL speed into warning indicator instead. I've implemented similar feature in my fork of MinimOSD - show remaining flight time given battery %, current draw, distance to home and RTL speed settings.

magicrub, 2016-10-16:
Does this PR satisfy this issue? https://github.com/ArduPilot/ardupilot/pull/5000

WickedShell, 2016-10-16:
It doesn't satisfy it for copter. It was a plane only, but it does pave the way for copter to be an easy addition.

WickedShell, 2018-03-10:
#7213 provides two stage battery failsafes, which you could configure low to be no action, which will still emit a warning message as well as set the battery warning bit to warn the user, then set the critical action to land. Closing this as we have more then enough issues open tracking this same feature set. (This is at least the 4th by my count)

closed 2018-03-10 by WickedShell