ArduPilot/ardupilot #3572: LAND is started when YAW in PosHold..where to disable?

AmsterdamHolland, 2016-01-31:
Hi everyone, 

I am new to Mission Planner.
 My quadcopter is flying OK, and all functionalities are working, but when I have it in Position Hold, and use Yaw to look around, almost immediately LAND or RTL is started ( not sure if it is LAND or RTL yet, have not flown since). 
 I taketh there must be a setting in Mission Planner where I could disable the LAND or RTL, so that I can yaw/look around freely.

Anyone who can steer me in the right direction?

I have the log file attached
[LogSettings.txt](https://github.com/diydrones/ardupilot/files/111558/LogSettings.txt)

I compared to another qaudcopter (the same, different owner) log file, the EKF is both set to 0,8, when turned to 0 it shouldn't Land? But I don't think this is wise, as the landing is not happening with this other quadcopter. It is the UPair Chase by the way. Thanks for the help.

AmsterdamHolland, 2016-02-05:
Can someone please help me?

I think the compass may be the culprit here, an offset of 300+ for Z others around 100.

Could this trigger the EKF check tresh fail safe?

rmackay9, 2016-02-05:
Looks like all the arming checks are off (ARMING_CHECK = 0).  That's a bad sign because it probably means some warnings are being ignored.

Is this an AVR board?  The compass orientation is "8" which is roll-180.  I believe that's correct if you're using an AVR board with an external GPS/compass module.  If it's a pixhawk then it's incorrect and should be set to "0".

Hard to say anymore without a dataflash log.

This is not really the place for support queries I'm afraid.  Sorry to be strict but this list is for confirmed bugs and enhancement requests and if we let support queries appear here it will just turn into a support forum and we won't be able to keep track of our to-do items for the next release.

closed 2016-02-05 by rmackay9

AmsterdamHolland, 2016-02-05:
Hi Randy, thanks for your reply.

I am sorry for posting this here, English is not my native language, I misunderstood, and thought I could post it here, my apologies.

It is the UPair Chase, with an external compass mounted upside down.
I have compared my log with another Chase owners log, and we both have the arming set to 0.
There was no data between our Chase that I found to be the cause.

Could my compass be faulty?
Can I set the EKF to 0 to be able to yaw in PosHold, without consequence if the Chase does not have other issues?

I would ask the company that build it, were it not that they have allegedly been taken over by a real estate tycoon, and for at least the coming weeks they will not be available via customer service.

Thanks again, I appreciate your help.