ArduPilot/ardupilot #3581: geofencing at the size of a country causes sim floating point overflow

marcmerlin, 2016-02-02:
I use a country-size fence because of setting min and max altitudes without a fence isn't possible currently ( https://github.com/diydrones/ardupilot/issues/2984 )
@tridge said that it was worth looking into while huge fences causes floating point errors in the sim though.

marcmerlin, 2016-03-29:
Hi Tridge, you asked me to file this. You can decide if it's still a problem, or not.

Naterater, 2019-01-07:
Just want to follow up on this as well if possible.

rmackay9, 2019-01-08:
I suspect the underlying floating point overflow does happen in Copter as well (so we should fix that) but at the same time Copter also allows setting up just an altitude fence so the work around to set a huge horizontal fence isn't required (for Copter).

amilcarlucas, 2021-03-09:
@joshanne I think your PRs #15568 and #15288 fixed this.
@marcmerlin can you test this with the latest master

marcmerlin, 2021-03-09:
Thanks for taking care of this. This bug was 5 years ago though, I'll be honest that I haven't been flying in a while, so unfortunately I can't test this, but since you've just fixed min/max altitude triggers without needing a country-wide fence, my guess is that this won't be an issue anymore.
Thanks for your work.

IamPete1, 2021-07-25:
Sounds like we can close this

closed 2021-07-25 by IamPete1