ArduPilot/ardupilot #3582: Plane: never pitch down to gain airspeed unless below FBW_ARSPD_MIN

magicrub, 2016-02-02:
When below the airspeed-trim and/or cruise airspeed, maintain airspeed with throttle and pitch up. Do not pitch down to speed up unless trying to prevent a stall. If you hit Throttle_max and still not reaching throttle trim and/or cruise speed then you've done the best you can and just carry on

WickedShell, 2016-02-03:
I don't think this is all that valid, there is speed weighting for this,
and this really related to the TECs needs anyways. Ie without knowing the
desired altitude climb behavior this could easily be incorrect. If I say I
want to prioritize airspeed then I should pitch down to get the airspeed.

marcmerlin, 2016-02-03:
@WickedShell: for my own info, can you explain a use case where an auto mode should decide to deviate from its programmed altitude and pitch down to gain airspeed for reasons other than stall avoidance?

marcmerlin, 2016-02-03:
@magicrub I would potentially even change your title to say "and stall prevention is enabled".
As a plane pilot, having an autopilot pitch down, really scares me, and unless it's to prevent a stall (stall warning enabled), it shouldn't even matter if I'm below FBW_ARSPD_MIN.
I would also argue that before pitching down, the AP should ensure that throttle is running at max allowed and only if that's the case, then pitch down.
Otherwise, pitching down, potentially towards a crash, if throttle isn't even at full power, seems like a bad choice (yes, there is some difficulty about deciding how long you wait after going to full throttle before deciding that full throttle isn't fixing the airspeed and pitch down is now the only option left to keep airspeed)

WickedShell, 2016-02-03:
I guess my concern is that if I have speed weight set to 1 I expect the plane to balance altitude against airspeed. If the target airspeed isn't achievable I expect the plane to do it's best at balancing those concerns. Getting all the way to its stall limit (even if we are attempting to stay above it) is not what I would expect to have happen. It's supposed to be evenly weighing the two. Especially since the stall prevention still allows 20 degrees of bank, meaning that to stay safe the FBW_MIN has to be artificially high anyways with this proposed change.

marcmerlin, 2016-02-03:
@tridge and I had a quick chat about it just 1h ago, and figured there shouldn't be opposition to this

If 
1) target altitude is higher than current altitude
2) airspeed is higher than FBW_MIN
then
airplane should never pitch down to gain airspeed as long as airspeed remains above FBW_MIN

This will catch the case of an engine failure during let's say RTL where the plane is supposed to climb, full power is applied but motor is dead, plane pitches up to main altitude, airspeed decays until FBW_MIN, and only then plane pitches down to maintain airspeed to avoid stall
Does that sound fair?

kwikius, 2016-02-11:
Was looking into slope soaring scenario ( no motor ) where wind speed is high and ground speed is low) Ideally would like to at least maintain position over the ground ( aka penetrate) at expense of height... Maybe this is not in Ardupilot scope though ?

magicrub, 2016-02-11:
airspeed is high so there's no problem. What are you concerned about?

kwikius, 2016-02-11:
Sorry I read "never pitch down to gain airspeed unless below FBW_ARSPD_MIN". Here is a situation where I must pitch down. Slope soarer.. no motor Only way to maintain speed over the ground in a strong wind is to pitch down

regards
Andy

marcmerlin, 2016-02-11:
@kwikius so you can pitch down if you're controlling pitch (FBWA/MANUAL), but this issue is to say that the auto pilot should not do this when it controls the pitch (FBWB/CRUISE/GUIDED/RTL).
This is because it is in control of keeping altitude, and not hitting the ground is a prime directive at this point.
You're correct that if you don't have a motor, and you RTL, you could end up in a place where you may not move forward unless you pitch down and lose altitude, but if the AP is flying, it's safer if it avoids hitting the ground instead of tries to keep some specific airspeed above FBW_MIN or ground speed.

magicrub, 2016-02-11:
Not stalling is a pretty high priority for the autopilot.

kwikius, 2016-02-12:
Without labouring it,  Motor or no motor ; Any time you wish to increase airspeed without gaining height then you must pitch down to lower the lift coefficient.

@Macmerlin. There is a distinction between RTL and other stab modes. In RTL priority is  to return to launch point.  In a strong wind if no progress is being made toward home, I would have thought trying to increase airspeed until/unless below the RTL specified height would be a useful course of action. Then look at strategy for why RTL is failing.. wind motor failure out of juice etc
@magicrub.  Pitching down (relative to earth frame)  will not lead to a stall:) Pitching down is the number 1 way to gain airspeed get some air moving over control surfaces for manoeuvrability and put yourself further from potential stall or tip stall. 

(tip local) stall is a result of trying to raise the lift coefficient beyond the point where (local) airflow breaks down. Therefore it would be simpler / more sensible just to say "**never allow the (local) lift coefficient to go above a specified maximum**"   This maxim holds true for any situation

( Mac permissible CL should ideally be looked up dependent on airspeed due to CL dependence on Reynolds number)

regards
Andy

marcmerlin, 2016-02-14:
@kwikius you are correct that sometimes you have to pitch down to increase airspeed, but all auto flightmodes have an assigned altitude already.
If you are above that altitude, they will already pitch down to get to their target altitude, so you will get what you want. Once that altitude is reached, they will stick to it, and pitching down to gain airspeed, unless you're about to stall, is not the right thing to do (unless it's directly pilot controlled, but then it's not an auto mode anymore).
In other words, @magicrub's original subject still stands in my opinion :)

magicrub, 2016-02-21:
@tridge care to comment on this?

marcmerlin, 2016-02-21:
@kwikius I think another issue is that you're talking about tall avoidance and this bug is not about this. This bug is about not pointing the nose towards a ground (and a possible crash) unless we have to, to avoid a stall.
As you may or may not know, you cannot properly predict a stall without an AOA (angle of attack indicator). Without that, most planes use airspeed as a next best indicator, but it can be off from actual stall speed by a fair amount.
So you're talking about CL, which is all nice and good, but without an AOA, you can't measure it, and either way it is not really relevant to this issue :)

So I refer you to these 3 issues instead;
https://github.com/diydrones/ardupilot/issues/2933
https://github.com/diydrones/ardupilot/issues/2669
https://github.com/diydrones/ardupilot/issues/2928

kwikius, 2016-02-22:
https://github.com/phollands discussed a way to estimate AOA here . https://github.com/diydrones/ardupilot/issues/2928#issuecomment-145323935

Dont see any argument to the fundamental principle that you must pitch down to gain airspeed if you wish to maintain same course, hence "never" is not valid. Seems like a blunt instrument. Was not the original issue due to an invalid set max airspeed?

Amount of pitch down is  proportional to how fast are going lower CL == lower angle of attack. . Seems weird that you should wait until airspeed is below min ( and going below min airspeed is more likely to happen since you have no guaranteed way to increase airspeed until you are below min airspeed)  IOW why not just say dont go below airspeed min and maintain height..  If you cant maintain height and keep airspeed above airspeed min then you must pitch down or stall :0 simples  :)  

And I think that is my whole issue with this. Its not a continuous function

Hey I am just going by the title, but overall I would like to see a somewhat more generic algorithm

magicrub, 2016-02-22:
in auto, it is always trying to maintain a cruise speed. Lets call it
15m/s. When below that, we propose increasing only throttle instead of
throttle+pitch. When below stall speed, lets call it 12 m/s, that means
throttle has been attempted already and is likely at max so at this point
we should allow pointing down to maintain it.

current method:
>=stall -> use throttle+pitchUp/Down to maintain cruise speed and altitude
< stall -> throttle+pitch down to ensure >= stall speed

proposed:
>=stall -> use throttle+pitchUp to maintain cruise speed and altitude
< stall -> throttle+pitchDown to ensure >= stall speed

WickedShell, 2016-02-22:
Tom: I really think there is some weighting on that and that that
oversimplifies the current behavior. I watched someone hand the autopilot
over when it was flying faster then cruise and at a lower altitude then it
wanted with motor output disabled. Because of the weighing of the two it
immediately traded off all of it's airspeed for altitude, then maintained
the minimum speed to the detriment of altitude quite nicely.

For anyone who's had an issue with this what was your TECS_SPDWEIGHT set
to?

kwikius, 2016-02-23:
Attached is a spreadsheet with a graph of function to make this functionality continuous FWIW, but I think it is accounted for already as @WickedShell  says...

//pitch_weight is weight to control by pitch , varying between 0 and 1
let power = choose[1 .. 100] ; // higher value gives sharper curve
pitch_weight = if ( air_speed < stall_speed, 1, max( 1 - ( air_speed - stall_speed) / (cruise_speed-stall_speed), 0) ^ power);

[pitchweight.ods.zip](https://github.com/diydrones/ardupilot/files/142345/pitchweight.ods.zip)

regards
Andy

marcmerlin, 2016-02-23:
@WickedShell I had this in https://github.com/diydrones/ardupilot/issues/3357 where TECS caused the plane to dive towards the ground towards a crash (I took over manually just before that happened).
From https://github.com/diydrones/ardupilot/files/67922/bfg6_3.4.param.txt
TECS_SPDWEIGHT,1.00000000
I don't think I ever touched it, I think that was the default.
Putting aside that the other bug stands, there is no reason to dive towards the ground to reach an airspeed not necessary for normal flight, should I adjust this parameter?

kwikius, 2016-02-23:
There is a function that looks like it is meant to address this problem in the AP_TECS code  
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.cpp#L673

regards
Andy

lekston, 2016-02-24:
Hi,

I would be careful with placing strong limits like the one mentioned in the title - sure
 let's just assume the two following scenarios:
  1. assume an airplane is in a steep climb and engine failure occurs, at this instant we are still well above the FBWA_ARSPD_MIN (an still below the desire altitude) but if we refrain from lowering the nose immediately, the airplane will most likely stall violently (the rate of change of airspeed is very high and waiting until FBWA_ARSPD_MIN will do no good)
  2. some airplanes have a strong positive pitch moment when throttle is rapidly increased (e.g. many highwing aircraft), so if speeds is close to (but higher than) FBWA_ARSPD_MIN and pitch is positive there is a risk of stalling just by setting max. throttle without pushing the nose to horizontal.

Sometimes, a pitch down of just a few degrees would allow the airplane to stay on the safe side.

As, for #3357, later comments in that topic suggest that the dive was related to very high setting ARSPD_FBW_MAX and the use of throttle nudge, so this seems unrelated.

Regards,
Przemek

marcmerlin, 2016-02-24:
@lekston so I didn't write the title, but I think "pitch down" meant having the nose pointed below the horizon.
I agree that the AP should of course be able to pitch down up to horizon level, regardless of the airspeed. Pitching below horizon if it's not trying to descend, is there ever a good reason to do this unless your speed is getting to/below the minimum flying airspeed?

WickedShell, 2016-02-24:
@marcmerlin It honestly sounds like the correct behavior still. There is a weighing between those, it will eventually yield the altitude to maintain the airspeed, and you're airspeed error was very high at the time. The aircraft shouldn't be allowed to be pushed into being closer to its stall envelope solely to maintain a given height. Indeed if I'm a bit high and slow then I should push the nose down and below to fix it, not just wait.

The biggest problem is that the proposed solution pushes us closer to stall territory in favor of just altitude, but that isn't actually safer, and depending on how close you're FBW min is set you would probably stall, and you definitely risk not getting a fast enough response to fix that (ie throttle response time might not be good, you could have just suffered a motor failure etc). In general this honestly just sounds like a case of having the parameters set incorrectly.

If you want to prioritize altitude that badly then lowering TECS_SPDWEIGHT towards 0 will give you the behavior. I wouldn't recommend it as the airspeed is a very real part of safe flight. I just don't see any reason why the aircraft shouldn't seek to minimize the error in it's speed/height, and at least putting the nose down for speed is the safer way to do so. (Less stall risk, responsive controls).

marcmerlin, 2016-02-24:
@WickedShell are you talking about my issue, namely https://github.com/diydrones/ardupilot/issues/3357 ?
If so, you cannot possibly say pitching down there was the right thing. Airspeed was way above stall, plane was below its minimum altitude deck, and it increased throttle and pitched down towards a crash to try and reach ARSPD_FBW_MAX without good reason to do so.
If this is what you're referring to, let's discuss on the other issue :)
More generally, we're talking on this issue about having safeguards when other things are already not going well. Telling someone who just crashed his $2000 plane that TECS_SPDWEIGHT, a fairly obscure option, was set wrong (i.e. not changed from its default) can't really be the right answer, correct? :)

@magicrub spun off this point from my issue I just referenced as a possible way to avoid very bad behaviour (i.e. diving towards the ground) when it's otherwise not necessary to do so.

Given all this, do you think there is a reasonable rule we can set so that if the autopilot is supposed to maintain altitude, it does not pitch down on its own to try and reach an airspeed that isn't necessary for flight?

WickedShell, 2016-02-24:
@marcmerlin it's balancing you're errors. You were reporting a very large error in airspeed, but low error in altitude. The concern is that this proposed solution throws away the ability to evenly balance the two errors which seems nonsensical. The problem still seems that the other parameter was wrong in the error you linked.

With reasonable ranges on airspeed I've been seeing really consistent and safe behavior which is why I'm opposed to this, the proposed solution would put me closer to an unsafe behavior, and is an awful combination for a windy day. Does that make sense?

marcmerlin, 2016-02-24:
@WickedShell can we please move to my issue if you're going to discuss it specifically?
There was no large error in airspeed. I just set the top achievable airspeed to a high value. Tridge looked into the logs and could not figure out why the AP decided that it had to achieve a higher airspeed and pitch down when the current airspeed was totally fine and GUIDED actually was supposed to climb.
But again, if you'd like to discuss this specific issue, let's move to it: #3357 

I do however not see why making the AP pitch below horizon level to gain airspeed when it's supposed to maintain altitude or climb, and well above the minimum flying airspeed, is something that is desirable, and wind is not relevant to this since it only affects ground speed (not talking about wind sheer).

Sorry if I'm being dumb, but please explain to me this:
1) airspeed > FBW_MIN   (or FBW_MIN + x% for extra safety)
2) AP is in some auto mode that commands keeping altitude or climbing and is at or below target altitude
3) AP is below some ideal airspeed commanded by TECS and not directly chosen by the pilot
4) AP decides to pitch down below horizon level to gain airspeed and lose altitude

How can #4 ever be the right thing to do given 1, 2, and 3?
Let's say I'm in FBWB, with altitude set at 100m. If I add throttle to go faster, are you saying it would be ok for the AP to decide to leave its target altitude and descend to go faster?
Now I'm in RTL or GUIDED, and again the AP is flying fine, well above FBW_MIN, but somehow decides that flying faster would be good, unless I'm above target altitude, would it ever be ok for the AP to pitch down and lose altitude, all the way to the crash site, just to gain airspeed not necessary for flight?

If we can't agree to this, how about you replace #2 with:
2) AP is below target altitude
AP needs to climb, airspeed is above FBW_MIN + margin, why would it ever pitch down and lose altitude?

WickedShell, 2016-02-24:
If that's the only way the AP can go faster then yes thats what I expect it to do. Airspeed is what keeps a plane alive. I never want to accept a perfect altitude at the expense of airspeed. The relative weighing of airspeed/altitude on a well setup plane should keep it safe. I've reviewed a log where the plane couldn't apply power and had enough altitude error that it very carefully ditched all of it's airspeed down to FBW_MIN trying to get the altitude back. It's about a balance between the two.

4 is the right thing to do if the airspeed is extremely far off from the desired, I'm not saying that it is to maintain perfect airspeed at the expense of altitude, but 4 is very valid if it's slow. There are response times on everything, and refusing to drop the nose until after we go below the minimum airspeed may be to late. And now my minimum airspeed needs to be set to my stall speed + a safety factor + safety factor for a 20 degree bank (since stall prevention wont kick in until then), and now I have to realize that I've sacraficed potential responsiveness by limiting bank angles to 20 degrees.

My point is that the hard rule of do not pitch down unless < min airspeed is not a smooth approach to applying inputs to the aircraft.

The current behavior does balance those, and with spdweight at the default value I've seen really reasonable response (10 meters low, 2 m/s slow all it did was add a large amount of throttle and lowered the pitch demand by 1 degree.

#3357 as I understood it was unreasonable that the small throttle nudge created such a large demand, but the execution of the demand seems reasonable. That's really all I was trying to say on it. (I would expect that at somepoint it would stop pitching down in favor of fixing the altitude error it gained, but that due to the relative amounts involved it just hadn't reached that).

marcmerlin, 2016-02-25:
@WickedShell I get your point the that original suggestion of airspeed > FBW_MIN  => do not pitch down is too strict.
Would you be ok doing this if airspeed > FBW_MIN + 40%?
If not what +x% would you be comfortable with?
To be clear, this is meant to be a last line of defense when other things are already going wrong.
Hopefully we can fix other issues that caused my almost big crash to happen, but I'm a defense in depth kind of person :)
(actually at work, if we get any catastrophic behaviour, we fix the problem that caused the behaviour to happen, but we also, whenever possible, add something to mitigate the impact should a similar situation happen again (like never delete more than 5% of all the users in one go, or something like that))
Is there a defense in depth that you'd be happy with here, as a percentage over FBW_MIN, or some other means?

marcmerlin, 2019-01-05:
sounds fair to me, thanks for the update